LDMX-Software/ldmx-sw #1392: [Discussion] Enforce python formatting?

tvami, 2024-08-15:
**Is your feature request related to a problem? Please describe.**

Now that we are enforcing c++ formatting rules, maybe it's time to think about Python formatting as well.

**Describe the solution you'd like**

Programs that are widely used and do this are 
 * Ruff: https://docs.astral.sh/ruff/
 * Black: https://github.com/psf/black

Note Ruff would have caught the issue about forgetting to `import os` that  triggered https://github.com/LDMX-Software/ldmx-sw/releases/tag/v4.0.3 and https://github.com/LDMX-Software/ldmx-sw/commit/640596b1541437faf1d9429901696de132127436
(sorry about that btw!)

**Describe alternatives you've considered**

Keep it the wild west!

tomeichlersmith, 2024-08-19:
I would love to include python formatting! It also opens the door to defining our own formatting rules for things like config scripts which could be helpful.

Personally, I like ruff since I'm a Rust fan girlie but black is also something I've used before and is pretty good.

tvami, 2024-08-19:
> Personally, I like ruff since I'm a Rust fan girlie 

hahaha! Sure if you have experience with `ruff` let's just do that!