LDMX-Software/ldmx-sw #1395: Change upstream beam location to come from the SP at the upstream TS

tvami, 2024-08-19:
**Is your feature request related to a problem? Please describe.**

Tracking points out that
```
 [ TruthSeedProcessor ] 3 : Linear extrapolation to a far away surface in B field.  This will cause inaccuracies in track parameters  Distance extrapolated = 5119.9
```
This is because now we shoot the beam from outside the `World`

**Describe the solution you'd like**

Let's shoot it from the upstream TS's SP

**Describe alternatives you've considered**

Defined another SP, but I think we agreed it's more physical to have it at the TS

tomeichlersmith, 2024-08-19:
I'll basically run the same analysis I did for https://github.com/LDMX-Software/SimCore/pull/98 (there is a tar-ball of my configs and notrbook) but use the scoring planes selecting on lowest z instead of the end position of the positron particle itself.

tomeichlersmith, 2024-08-19:
Running backwards positrons (with detector elements removed) and selecting the furthest upstream TS scoring plane by ID and only hits in that scoring plane from the primary positron (trackID 1), we get the following table.

This seems to make sense to me, so I'm going to copy these in.

beam|x|y|z|px|py|pz
---|---|---|---|---|---|---
4|-43.56748|0.0|-882.5055|388.5554|-0.0|3981.5967
8|-21.745876|0.0|-882.5055|388.55154|-0.0|7991.0703
1.2|-148.95303|0.0|-882.5055|388.57147|-0.0|1135.8867

PR #1396 (2024-08-20): copy in positron-deduced positions and directions from the TS scoring planes
moving the z-position up to the upstream face of the furthest upstream trigger pad so that it is more feasibly understood by the Tracking algorithms

I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?

This resolves #1395 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.

[image]

The tar-ball below has the configs I used and the python scripts I used to extract the beam po [... cut]