LDMX-Software/ldmx-sw #1402: Tracking timing changes in log diffs

tvami, 2024-08-22:
**Is your feature request related to a problem? Please describe.**
Logs always fail now because of the character count different
```
<  [ SeedFinderProcessor ] 1 : AVG Time/Event: 0.0986296 ms
  ---
  >  [ SeedFinderProcessor ] 1 : AVG Time/Event: 0.0978154 ms
  244c244
  <  [ SeedRecoil ] 1 : AVG Time/Event: 0.136037 ms
  ---
  >  [ SeedRecoil ] 1 : AVG Time/Event: 0.133573 ms
  256c256
  <  [ Tagger_TrackFinder ] 1 : AVG Time/Event: 0.395085 ms
  ---
  >  [ Tagger_TrackFinder ] 1 : AVG Time/Event: 0.39516 ms
  258,263c258,263
  <  [ Tagger_TrackFinder ] 1 : setup       Avg Time/Event = 0.00228943 ms
  <  [ Tagger_TrackFinder ] 1 : hits        Avg Time/Event = 0.0433836 ms
  <  [ Tagger_TrackFinder ] 1 : seeds       Avg Time/Event = 0.00342895 ms
  <  [ Tagger_TrackFinder ] 1 : cf_setup    Avg Time/Event = 0.00407709 ms
  <  [ Tagger_TrackFinder ] 1 : ckf_run     Avg Time/Event = 0.000160371 ms
  <  [ Tagger_TrackFinder ] 1 : result_loop Avg Time/Event = 0.339842 ms
  ---
  >  [ Tagger_TrackFinder ] 1 : setup       Avg Time/Event = 0.00216297 ms
  >  [ Tagger_TrackFinder ] 1 : hits        Avg Time/Event = 0.044[55](https://github.com/LDMX-Software/ldmx-sw/actions/runs/10496314488/job/29077738861?pr=1401#step:6:58)07 ms
  >  [ Tagger_TrackFinder ] 1 : seeds       Avg Time/Event = 0.00345044 ms
  >  [ Tagger_TrackFinder ] 1 : cf_setup    Avg Time/Event = 0.00419966 ms
  >  [ Tagger_TrackFinder ] 1 : ckf_run     Avg Time/Event = 0.0001725 ms
  >  [ Tagger_TrackFinder ] 1 : result_loop Avg Time/Event = 0.33878 ms
  265c265
  <  [ Recoil_TrackFinder ] 1 : AVG Time/Event: 0.298821 ms
  ---
  >  [ Recoil_TrackFinder ] 1 : AVG Time/Event: 0.308164 ms
  267,272c267,272
  <  [ Recoil_TrackFinder ] 1 : setup       Avg Time/Event = 0.000543621 ms
  <  [ Recoil_TrackFinder ] 1 : hits        Avg Time/Event = 0.042665 ms
  <  [ Recoil_TrackFinder ] 1 : seeds       Avg Time/Event = 0.00260774 ms
  <  [ Recoil_TrackFinder ] 1 : cf_setup    Avg Time/Event = 0.00199831 ms
  <  [ Recoil_TrackFinder ] 1 : ckf_run     Avg Time/Event = 0.000136034 ms
  <  [ Recoil_TrackFinder ] 1 : result_loop Avg Time/Event = 0.250555 ms
  ---
  >  [ Recoil_TrackFinder ] 1 : setup       Avg Time/Event = 0.000690733 ms
  >  [ Recoil_TrackFinder ] 1 : hits        Avg Time/Event = 0.0449752 ms
  >  [ Recoil_TrackFinder ] 1 : seeds       Avg Time/Event = 0.00296358 ms
  >  [ Recoil_TrackFinder ] 1 : cf_setup    Avg Time/Event = 0.00209514 ms
  >  [ Recoil_TrackFinder ] 1 : ckf_run     Avg Time/Event = 0.000153931 ms
  >  [ Recoil_TrackFinder ] 1 : result_loop Avg Time/Event = 0.2[56](https://github.com/LDMX-Software/ldmx-sw/actions/runs/10496314488/job/29077738861?pr=1401#step:6:59)973 ms
```

**Describe the solution you'd like**

We should round these numbers to a fixed number of digits.
But also it's interesting that rerunning ACTS on the same events takes different times.

**Describe alternatives you've considered**

Figure out why those numbers are different, is that expected @bloodyyugo 

**Additional context**

PR test here https://github.com/LDMX-Software/ldmx-sw/actions/runs/10496314488/job/29077738861?pr=1401

tomeichlersmith, 2024-08-22:
I mean, the times are ranging from tenths to hundreds of _microseconds_ so I feel like the time changing is just a product of the performance of the runner that we happen to land on.

tvami, 2024-08-22:
Sure that makes sense, let's just round them and get the diffs to zero with that solution!

closed 2024-09-13 by tvami