LDMX-Software/ldmx-sw #1405: Error in <TDecompSVD::Diagonalize>: no convergence after XX steps

tvami, 2024-08-23:
**Describe the bug**

This msg from the EcalVeto has been there since forever, and is probably not a big deal
```
Error in <TDecompSVD::Diagonalize>: no convergence after XX steps
```

**To Reproduce**

Just run the ECAL veto.

**Desired behavior**

Not to have this error.
My thinking is that it's having a matrix that has repeated elements, and the fix is probably just about conditioning what goes into the `TDecompSVD`. I'll have a look at it in details and report back.

EinarElen, 2024-08-24:
I think that if you fix this issue, you will fix a large part of why the ecal veto processor is so much slower than all of the other eco-processors :)

tvami, 2024-08-25:
Seems like a pretty obv case, here are a few examples where it works, and one when it does nt
```
----  hdt: 
     |12.0397 | -6.9511 | 0 | 
     |-2.40793 | 1.39022 | 0 | 
     |-9.63173 | 5.56088 | 0 | 
 decomposed : 1
 ----  hdt: 
     |2.40793 | -1.39022 | 0 | 
     |2.40793 | -1.39022 | 0 | 
     |-4.81586 | 2.78044 | 0 | 
 decomposed : 1
 ----  hdt: 
     |9.63173 | 0 | 0 | 
     |9.63173 | 0 | 0 | 
     |-19.2635 | 0 | 0 | 
Error in <TDecompSVD::Diagonalize>: no convergence after 31 steps
Error in <TDecompSVD::Diagonalize>: no convergence after 32 steps
 decomposed : 0
 ----  hdt: 
     |12.0397 | -1.39022 | 0 | 
     |4.81586 | 2.78044 | 0 | 
     |-16.8555 | -1.39022 | 0 | 
 decomposed : 1
 ----  hdt: 
     |-2.40793 | 1.39022 | 0 | 
     |4.81586 | -2.78044 | 0 | 
     |-2.40793 | 1.39022 | 0 | 
 decomposed : 1
 ----  hdt: 
     |67.601 | 44.797 | -135.765 | 
     |-30.1886 | -24.4838 | 67.8827 | 
     |-37.4124 | -20.3132 | 67.8827 | 
 decomposed : 1
 ----  hdt: 
     |74.8248 | 46.1872 | -135.765 | 
     |-22.9648 | -23.0936 | 67.8827 | 
     |-51.86 | -23.0936 | 67.8827 | 
 decomposed : 1
 ----  hdt: 
     |77.2328 | 44.797 | -135.765 | 
     |-27.7807 | -20.3132 | 67.8827 | 
     |-49.4521 | -24.4838 | 67.8827 | 
 decomposed : 1
 ----  hdt: 
     |24.2583 | 47.5774 | -135.765 | 
     |-8.51723 | -25.874 | 67.8827 | 
     |-15.741 | -21.7034 | 67.8827 | 
 decomposed : 1
```

closed 2024-08-29 by tvami