LDMX-Software/ldmx-sw #1406: Check for corrupted / zombie files in ldmx-sw?

tvami, 2024-08-24:
**Is your feature request related to a problem? Please describe.**

This https://github.com/LDMX-Software/LDCS/issues/23 made me think if we should have a built-in check for input files being corrupted / zombie.

**Describe the solution you'd like**

I'd think here
https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Framework/src/Framework/Process.cxx#L279
is where this should go, but I dont know enough about the `EventFile` class, and a quick look almost suggests that it's like a `TFile` wrapper. So this makes me think if we could do 

```
if (!inFile || inFile->IsZombie()) continue;
```

tomeichlersmith, 2024-08-28:
`EventFile` does not inherit from `TFile` but implementing a quick `isCorrupted()` function for it would make sense.

We actually already check for existence of both the file and the event tree in the EventFile constructor.

https://github.com/LDMX-Software/ldmx-sw/blob/d74f8ccc86524fb3de6fbc89ef70033ee6c61b73/Framework/src/Framework/EventFile.cxx#L53-L69

Perhaps we include a check for `IsZombie` in there as well?

Personally, I really don't like the idea of continuing smoothly past a corrupted file. I think the Framework should end the program if a corrupted file is encountered in order to inform the user of this corruption. Maybe we can get around the failing-on-the-last-file annoyance by making sure to `Write` the output file at the end of every input file?

tvami, 2024-08-29:
> I really don't like the idea of continuing smoothly past a corrupted file. 

Well for the use case I ran into this, this would be not great. I had to ask LK to delete the input file and until they did it, I couldn't move on. (My use case was making a BDT skim, where I have 50 input files per output file)

So continuing on the bad file, and reporting that it was bad in the end of the workflow would be my preferred solution.

tvami, 2024-08-29:
> Rucio and LDCS - is it possible to mark files as bad and re-run a workflow so that bad files are skipped?

My understanding from @bryngemark is that this cannot be done.

> This might be a case where we add another flag to the constructor about "strictness"

Yes another flag could work, so in default we'd exit if the file is problematic, but if the person doing this knows about the flag they can let the code skip files (the pool for people who use LDCS is anyway pretty limited)

bryngemark, 2024-09-02:
That's right, there is currently no smooth way out of this with LDCS. What we do is specify an input data set, and a number of files per job, and then `aCT` queries the `rucio` database for actual file names and paths and chooses which files a given job will run on. Resubmission of a failed job (yet to be fully implemented in a completely useful way) would repeat the same job with the same input files. `aCT` does not know about the reason for the job failure. 

A flag sounds like a sensible solution to me, and would be hugely useful when we run harsh skims. For book-keeping, though, it would be great if the number of successfully opened files could be reported somewhere the metadata harvesting for `rucio` could find it. Otherwise, we'll soon be lost when it comes to EOT calculations.

PR #1443 (2024-09-06): check for corrupted zombie files in ldmx sw
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1406 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.

## Necessary Input Files
- `buggieFileKaon.root`: corrupted zombie file found by Tamas
- `justSim_10_events.root`: not corrupted, simulated file created by SimCore/test/basic.py

<details>

  <summary>merge-cfg.py</summary>

```python
from LDMX.Framework import ldmxcfg
p = ldmxcfg.Proce [... cut]
commit a113e1b5 (2024-09-06): check for corrupted zombie files in ldmx sw (#1443)

This resolves #1406 

On trunk, the current behavior is to try to continue even after a known
corrupted file is found. This leads to wacky results. The terminal
output is very long and contains a lot of detail about errors including the
branches that are struggling to be copied into the output file. 

Even more confusingly, some events end up in the final output file but
the run does not. These events could contain data that is corrupted? I am
not sure and have no means of checking.

This commit updates the Framework to end processing if a c [... cut]
closed 2024-09-06 by tomeichlersmith