LDMX-Software/ldmx-sw #1409: Introduce verbosity flag for tracking modules

tvami, 2024-08-25:
**Is your feature request related to a problem? Please describe.**

I'd like to have the CI test for the ecalPN back to the debug log level, but the tracking module is very verbose.

**Describe the solution you'd like**

I propose to introduce an extra `verbose_` booleans and have hit level information under it, and have it False by default. Anything that's track or event level should stay as it is, and `verbose_` should be configurable, so it's easy to have it back. (We have a similar approach in the EcalVeto already)

Some places where I'd propose to have this
- https://github.com/LDMX-Software/ldmx-sw/blob/a97526d09d370a6b99fc4bb49f743c60048709a5/Tracking/src/Tracking/Reco/GSFProcessor.cxx#L254
- https://github.com/LDMX-Software/ldmx-sw/blob/a97526d09d370a6b99fc4bb49f743c60048709a5/Tracking/src/Tracking/Reco/DigitizationProcessor.cxx#L234-L237
- https://github.com/LDMX-Software/ldmx-sw/blob/a97526d09d370a6b99fc4bb49f743c60048709a5/Tracking/src/Tracking/Reco/CKFProcessor.cxx#L486-L498
- Also this could be in one line instead of 8 lines https://github.com/LDMX-Software/ldmx-sw/blob/a97526d09d370a6b99fc4bb49f743c60048709a5/Tracking/src/Tracking/Reco/TruthSeedProcessor.cxx#L294 

**Describe alternatives you've considered**

There is an issue that discusses per module logging level setting (I'm not finding it rn, I'll add later), but until that's developed this is a solution.

tvami, 2024-08-25:
@bloodyyugo I'd love to hear what's your take on this! If you agree I'm happy to do the development, maybe after we moved to v36 ACTS.

bloodyyugo, 2024-08-26:
Yeah, I agree it's way too verbose for folks not looking to debug tracking.  I guess the easiest thing is having a  processor-level debug.  There already are TWO for CKFProcessors, debugTransform and debug (which sets the ACTs debug level).  That is a bit messy already.  

Either of the two options you mention above seem good to me.

tvami, 2024-08-26:
> There already are TWO for CKFProcessors

oh I didnt realize that haha! Well then the one in [CKFProcessor.cxx here](https://github.com/LDMX-Software/ldmx-sw/blob/a97526d09d370a6b99fc4bb49f743c60048709a5/Tracking/src/Tracking/Reco/CKFProcessor.cxx#L486-L498) could actually be under that flag.

Honestly I think the module level logging setting would be a better solution but I dont really know enough about the framework to implement that. Maybe @tomeichlersmith can comment.

tomeichlersmith, 2024-08-26:
We use Boost.Log to handle our logging. The biggest complexity is just the logic on how to filter logging messages in an _efficient_ way.

We currently only have one filter and it is the fastest possible (non-trivial) filter - just one integer comparison. If we wanted to have processor-level log-levels[^1], we'd need a filter to check both the severity attribute and some other attribute of the message (probably `"Channel"` which is where I put the name of the processor being run. ( :warning: The name of a processor and its class name are different strings). Writing this in an efficient way is something that isn't impossible and may not be a big enough performance impact for us to notice anyways.

https://github.com/LDMX-Software/ldmx-sw/blob/a97526d09d370a6b99fc4bb49f743c60048709a5/Framework/src/Framework/Logger.cxx#L89-L90

[^1]: You're CMSSW is showing lmao ;) In LDMX, I've gotten in the habit of calling each processor class a processor so that a group of them can be a module (e.g. CKFProcessor is an example of a processor and Tracking is an example of a module). I know CMS has a slightly altered vocabulary, just want to be clear here. If you are genuinely interested in applying the same log-level to an entire ldmx-sw module, we could do that, we'd just need to add a new attribute to the messages that includes what module they originate from.

tvami, 2024-08-26:
> You're CMSSW is showing lmao

hahah sorry!

OK, so the "real" module-level would be interesting too, but seems like that comes with way to much change (I understand every single ldmx_log  would need to be rewritten). Then for the processor-level, so this seems like a somewhat bigger project then what this issue brought up, so maybe the plan of action is to have the verbose flags in tracking, and then go into the logger development.

tomeichlersmith, 2024-08-26:
TLDR: We would *not* need to rewrite each `ldmx_log`. All of the filtering and attribute definition can happen during set up. (either in the `logging::open` or `logging::makeLogger` function.

---

Boost.Log (like many other loggers) are designed to avoid re-writing logging messages as much as possible. With this in mind, the messages are directed to "loggers" which then are what interact with "sinks" which can have various "filters" applied. If we want to add the event number to the output, we'd probably need to add an "attribute" to loggers so that information can be passed to the sinks (update `logging::makeLogger` and probably add something in `Process::run` pass the event number to the loggers). If we just want to set log levels (a.k.a. "Severity") depending on the processor (a.k.a. "Channel"), then we just need to define more intricate filters for the sinks to apply whenever they receive a message (change `logging::open`).

tvami, 2024-08-29:
Closing this in favor of https://github.com/LDMX-Software/ldmx-sw/issues/922

closed 2024-08-29 by tvami