LDMX-Software/ldmx-sw #1419: Double check secondary corrections for ECAL digi

tvami, 2024-08-29:
**Is your feature request related to a problem? Please describe.**

I was just talking with Tom regarding these numbers
https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Ecal/python/digi.py#L207
seem to be specific for 4 GeV.

**Describe the solution you'd like**

Just check if the numbers are very different for 8 GeV or not.
I understand from Tom that the way these were calculated is simply to shoot an 8 GeV electron gun into the ECAL and fit a double-sided Gaussian

**Describe alternatives you've considered**

It's second order effect, probably not too high of a priority

tvami, 2024-08-30:
@anandaguha can you please look into this? You are running guns anyway

anandaguha, 2024-08-31:
I am just not 100% sure about what distribution I will be fitting the double sided Gaussian to, but everything else seems pretty simple for me to do.

tvami, 2024-08-31:
> sure about what distribution I will be fitting

that ok, let's chat about it!

closed 2025-01-13 by tvami