LDMX-Software/ldmx-sw #1433: ECAL pn with no ECAL geometry?

tvami, 2024-09-04:
**Describe the bug**

This is more of a question / something to look into in details, but if I run the ecal pn with `ldmx-det-v14-8gev-no-cals` it produces events. A little bit intuitive given that there is no ECAL volume to have the ECAL pn to happen on. 

**To Reproduce**
Take https://github.com/LDMX-Software/ldmx-sw/blob/trunk/.github/validation_samples/ecal_pn/config.py
and modify the geometry to `ldmx-det-v14-8gev-no-cals` 

**Desired behavior**

This might be somehow all fine, but one should run the stepping function to see what's happening.

**Additional context**

Geom introduced in https://github.com/LDMX-Software/ldmx-sw/pull/1424/

EinarElen, 2024-09-04:
I think a thorough review and overhaul of the filtering system is long overdue, they keep causing issues and are hard to understand. And they are a perf issue #1376

tvami, 2025-04-08:
I think this is gone now, closing it!

closed 2025-04-08 by tvami