LDMX-Software/ldmx-sw #1455: Multiple printouts when using logFrequency

tvami, 2024-09-14:
**Describe the bug**

With the settings of
```
p.logFrequency = 100
p.termLogLevel = 1
```
I'd expect a single msg every 100 event to show up, instead, every 100 event the following
```
[ Process ] 1 : Processing 100 Run 2 Event 100  (2024-09-13 17:11:04.004336000-0700)
[ Process ] 1 : Processing 100 Run 2 Event 100  (2024-09-13 17:11:04.005039001-0700)
```
this comes up >5000 times!

**To Reproduce**
I was running this
```
from LDMX.Framework import ldmxcfg
p=ldmxcfg.Process("v14_nonfid")

from LDMX.Ecal import EcalGeometry
from LDMX.Hcal import HcalGeometry
import LDMX.Ecal.ecal_hardcoded_conditions as ecal_conditions
import LDMX.Hcal.hcal_hardcoded_conditions as hcal_conditions
from LDMX.Biasing import ecal
from LDMX.SimCore import generators

mysim = ecal.nonfiducial_photo_nuclear('ldmx-det-v14-8gev', generators.single_8gev_e_upstream_tagger())
mysim.description = "ECal Non-Fiducial Test Simulation"

#from LDMX.Biasing import util
#mysim.actions.append( util.StepPrinter(1) )

import LDMX.Ecal.digi as ecal_digi
import LDMX.Ecal.vetos as ecal_vetos

p.outputFiles = [f'events_nonfiducial_test_production.root']
p.histogramFile = f'hist_nonfiducial_test_production.root'

p.maxTriesPerEvent = 10000
p.maxEvents = 100
p.totalEvents = 1000
p.run = 2
p.logFrequency = 100
p.termLogLevel = 1

p.sequence=[ mysim,
        ecal_digi.EcalDigiProducer(),
        ecal_digi.EcalRecProducer(),
        ecal_vetos.EcalVetoProcessor()
        ]

from LDMX.DQM import dqm
p.sequence.extend(dqm.ecal_dqm)
```

**Desired behavior**

Just have that print out once after ever N event

**Environment:**
```
denv_workspace="/home/vamitamas/NonFiducialSimu/v4.1.2"
denv_name="v4.1.2"
denv_image="ldmx/pro:v4.1.2"
denv_mounts="/home/vamitamas/NonFiducialSimu/v4.1.2 "
denv_shell="/bin/bash -i"
denv_network="true"
apptainer version 1.2.5
```

tomeichlersmith, 2024-09-16:
This is expected since the outer loop over number of events is separate from the inner loop on the number of trials. We printout conditioned on the event index (decided by the outer loop) which means if there are multiple trials for an event matching the frequency, multiple printouts will be made.

Multiple printouts is confusing, but I'm curious what an improvement would be. Would just add a check that skips printouts if we are on later trials?

tvami, 2024-09-16:
> Would just add a check that skips printouts if we are on later trials?

Yes, I think that would be sufficient and satisfactory

PR #1458 (2024-09-17): logging fix and the long-awaited setting log-level by processor
These developments do the following
- interpret the message's logging level into a human readable string for easier parsing
- add event number to logging messages via a custom Formatter which resolves #1301 
- allow logging levels to differ between different channels by implementing a custom Filter which resolves #922 
- update the configuration so that the logging has its own configuration class, keeping its parameters organized and separate from the Process (the current syntax is supported by overriding `__getattr__` so this doe [... cut]
closed 2024-09-17 by tomeichlersmith