LDMX-Software/ldmx-sw #1457: Auto-commit clang format

tvami, 2024-09-14:
**Is your feature request related to a problem? Please describe.**

I know we spoke about this but I forgot the limitation, so let's persist this discussion in an issue. We all have forgetten to clang format at one point.

**Describe the solution you'd like**

It would be really nice if the CI would just do it for us.

Reading here
https://stackoverflow.com/questions/57921401/push-to-origin-from-github-action/58393457#58393457
this should be possible with the `run` command. And then it brings up this question of who should be the author but then it could just be the bot itself
```
git config user.name 'github-actions[bot]'
git config user.email 'github-actions[bot]@users.noreply.github.com'
```

tomeichlersmith, 2024-09-16:
The original limitation preventing this was pushing to submodules but that isn't an issue anymore.

Now I think the limitation is softer - we have a lot of developers who are not familiar with `git` and thus having commits created on their branches for them could be a source of confusion. This could be resolved by some helpful documentation and guidance on slack, but I think this is something I want to consider.

tvami, 2024-09-16:
Yes I agree, I think this is worth doing it, also good education for people regarding git magic :D

closed 2024-09-18 by tvami