LDMX-Software/ldmx-sw #1468: Change trigger threshold defaults

tvami, 2024-09-19:
**Is your feature request related to a problem? Please describe.**

https://github.com/LDMX-Software/ldmx-sw/blob/c1164a4452c210840cec065115b2b3bcecbf62ed/Recon/python/simpleTrigger.py#L43
```
 self.thresholds = [ self.beamEnergy/4000.*1500.0, self.beamEnergy/4000.*1000. + self.beamEnergy, self.beamEnergy/4000.*500. + 2*self.beamEnergy, self.beamEnergy/4000.*100. + 3*self.beamEnergy ]
```
leads to 3000, 10000, 17000, 24200, while
@EBerzin 's studies lead to different numbers
https://indico.fnal.gov/event/64932/#25-multi-electron-trigger-thre
(although I think Elizabeth was looking at the 34 layers of ECAL, which is not the same as the one we have )
 
**Describe the solution you'd like**

Agree on the final numbers and have them in ldmx-sw.
I should add that I strongly believe we should keep the 1e threshold at 3000, so we dont need to redo all the 1e studies (trig skim, BDT, etc), even if it's not fully optimal.

EBerzin, 2024-09-19:
I've attached the full set of optimized trigger thresholds (20 and 34 layers for mu = 1 and mu = 2). For 20 ECAL layers, the thresholds are 3.01, 10.79, 18.54, 26.25 GeV. So keeping the 1e threshold at 3000 is consistent with this. 

[LDMX_TS_Trigger_Thresholds.pdf](https://github.com/user-attachments/files/17067444/LDMX_TS_Trigger_Thresholds.pdf)

tvami, 2024-09-19:
Awesome, thanks a lot @EBerzin !
So then next is to figure out how to implement this so it works for both 4 GeV and 8 GeV. Maybe just add an if on the beamEnergy hand have two different lines for `self.thresholds` as opposed to having it as a function of the energy? Elizabeth do you wanna go ahead with the PR, or should I?

EBerzin, 2024-09-20:
I can take of the PR.

closed 2024-09-27 by tomeichlersmith