LDMX-Software/ldmx-sw #1481: Add pass name to tracking code (CKFProcessor, Vertexing, etc)

tvami, 2024-09-30:
**Is your feature request related to a problem? Please describe.**

Filippo was running ACTSv35 on an existing eN sample where the tracking branches were already created and ran into the issue of
```
[ fire ] 1 fatal: [ProductAmbiguous] : Multiple products found for name 'RecoilTruthSeeds' without specified pass name :
{ name = RecoilTruthSeeds, pass = genie_reco, type = vector<ldmx::Track>}
{ name = RecoilTruthSeeds, pass = genie_reco_v36, type = vector<ldmx::Track>}
  at /Users/fdelzanno/Desktop/Incandela/Coding/ldmx-sw/Framework/include/Framework/Event.h:300 in getObject
```

**Describe the solution you'd like**

this
https://github.com/LDMX-Software/ldmx-sw/blob/34d43a553320a549b81a9dd30f31ab7c826a0d93/Tracking/src/Tracking/Reco/CKFProcessor.cxx#L239
```
 event.getCollection<ldmx::Track>(seed_coll_name_);
```
should be
```
 event.getCollection<ldmx::Track>(seed_coll_name_, input_pass_name_);
```
probably true for the vertexer and other processors in the tracking.

**Describe alternatives you've considered**

Dropping the old collection and running on the new one in two steps. But this wont allow comparisons in one config for sure

bryngemark, 2024-09-30:
good catch, configurable input collection and pass names, and output collection names, should be standard for all processors in my opinion

tvami, 2024-09-30:
Yes, I agree!

tvami, 2025-02-17:
Resolved in https://github.com/LDMX-Software/ldmx-sw/pull/1607

closed 2025-02-17 by tvami