LDMX-Software/ldmx-sw #1493: Add simulation level "trigger filter"

tvami, 2024-11-13:
**Is your feature request related to a problem? Please describe.**

From 1M inclusive events about 100 will pass the trigger, if we are interested in looking at inclusive events after trigger we really need a lot of stats.

**Describe the solution you'd like**

We discussed at today's software meeting the possibility of having a filter that would enhance the triggering efficiency. This would mean to add the energy in the ECAL volume and if it's above a threshold the event would be not saved. OTOH we didnt come up with a concrete solution, so this issue serves also as a planning board.

**Describe alternatives you've considered**

If the filter is computationally too expensive it makes sense to just generate much bigger amounts of inclusive events and trigger skim

**Additional context**

This would be relevant for several projects: tracking studies, multielectron studies, rare physics events checks (e.g. WABs)

tomeichlersmith, 2024-12-05:
This is a large lift because it is difficult to define "energy in the ECal volume". In other filtered samples (e.g. the Ecal-as-Target samples), we choose a process to filter for. This makes the energy sum much easier (but even when requiring > 60% of the beam energy to go into photon-nuclear processes, the majority of events fail the trigger). We _could_ have a simulation filter which selects events based on the amount of energy deposited in the ecal sim hits, but we would then need to wait to apply that selection until after the entire shower is simulated anyways which is very close (in time since emulation/reconstruction/trigger is so fast) to just using the trigger processor itself.

What I've done at UMN is to do the trigger skim at the same time as the simulation. This does not avoid the CPU-time burden of simulating so many events that are then thrown out, but it does save all of that disk space (and the time necessary to write/read all of that data). Making sure the trigger skim is properly configured is even easier than making sure a simulation filter is because you can just write out all events (comment out the `p.skimConsider` and `p.skimDefaultIsDrop` lines) for a few runs and double check the trigger decision is what you would expect it to be.

tvami, 2024-12-10:
Cant we just look at the electron+photon (or whatever else is produced) energy in the step that's after the target, and if it's more than a threshold energy abort the event? There is no way that a 5 GeV electron + 2 GeV photon will have a shower that's less than 3 GeV, right? (Well I assumed the electron is fiducial in ECAL). This wont be perfect of course, but right now the inclusive to trigger selection is orders of magnitude change, so if with this requirement we can get something that 50% I think it's worth it. I guess one question is would this approach miss events that would otherwise pass the trigger.

tvami, 2024-12-10:
I mean even having a max energy filter on the electron would reduce a lot.

tomeichlersmith, 2024-12-10:
I think this strategy would work well for studying the "most normal" events that could pass the trigger. The problem is there are also a lot of events where just something weird happens in the ECal and less energy is reconstructed.

> no way that a 5 GeV electron + 2 GeV photon will have a shower less than 3 GeV

There is! In my EaT studies, I have a 1B EoT sample requiring the electron to reach the ECal with >7GeV and we still see O(few hundred) pass the trigger requirement (PN and DiMuon happens in the shower). Requiring the total energy to be less than 7GeV would certainly have a higher rate of events passing the trigger but that sample would also miss these kinds of events where a rare process just happens early enough in the shower or not enough energy is deposited in the few sensitive layers.

In some sense, this is why I am advocating for just doing the trigger processor if what you want is an _inclusive_ sample. I cannot think of a filter than can properly sample all of the weird events from the tails that pass the trigger.

If you are okay with not having an inclusive sample and just want some trigger-able events to study, then I think a selection on the particles leaving the target makes sense. It's just important to note that the sample would not be truly inclusive even though we are avoiding biasing a specific process.

tvami, 2024-12-18:
>  the electron to reach the ECal with >7GeV and we still see O(few hundred) pass the trigger requirement (PN and DiMuon happens in the shower)

I'm trying to digest this. Did you mean that PN and the DiMuon is coming from the brem photon? What's happening with the electron in these events? It reaches the ECAL face with >7 GeV and then makes a shower and then in the shower the particles are just not ionizing enough?

tvami, 2024-12-23:
ok, thanks Tom for your input, I guess this can't be done (efficiently) and I'll "close this as not planned" then

closed 2024-12-23 by tvami