LDMX-Software/ldmx-sw #1502: Add realistic read-out threshold in EcalVeto

tvami, 2025-01-03:
**Is your feature request related to a problem? Please describe.**

Here
https://github.com/LDMX-Software/ldmx-sw/blob/96f698064f435dddc2b6f1290bf81aab5b492c9d/Ecal/src/Ecal/EcalVetoProcessor.cxx#L395
```
    if (hit.getEnergy() > 0) {
```
all the hit related quantities have no threshold. That doesnt seem very realistic.

**Describe the solution you'd like**

Add a threshold value, maybe have it configurable since the read-out threshold could change.

**Describe alternatives you've considered**

Do nothing, it's probably a small effect?

**Additional context**

This question came up in the DR review, regarding the definition of "hit" in the plots.

tvami, 2025-02-02:
A similar way, so I'm not opening another issue about it

I think the 
```
totalRecEnergy += recHit.getEnergy();
```
should go before the 
```
skip anything that digi flagged as noise
```
in https://github.com/LDMX-Software/ldmx-sw/blob/trunk/DQM/src/DQM/EcalDigiVerifier.cxx#L65-L69

@tomeichlersmith do you agree? (Seems cheating since we wont actually know in real data if it's genuine noise or not)

tomeichlersmith, 2025-02-03:
I also don't think we should check if its noise.

I can't find where/if I've said this before, but I don't think there should be a readout threshold in the EcalVeto at all. If there are hits being reconstructed with negative energy, I would consider that a bug in the reconstruction or emulation and not something to be filtered out at the EcalVeto. There should be a readout threshold at the emulation and maybe a secondary one at the reconstruction but not at the EcalVeto.

tvami, 2025-02-03:
OK, I agree it's more of a digitization question than a "veto question". 
But we dont have a threshold in the digi, do we?

tomeichlersmith, 2025-02-03:
We do

https://github.com/LDMX-Software/ldmx-sw/blob/2d449dc907f3b131b1c3c48c575695d0eae795f0/Tools/src/Tools/HgcrocEmulator.cxx#L216-L218

but if its not working correctly and letting some negative-energy (i.e. below-pedestal) hits through, then we need to debug the emulation.

closed 2025-02-10 by tvami