LDMX-Software/ldmx-sw #1505: just configure fails if denv workspace exists above ${LDMX_BASE}

Skelpdar, 2025-01-06:
**Describe the bug**
`just init` does not initialize a new denv workspace properly, if a workspace already exists further up the file hierarchy.

**To Reproduce**
The file structure looks something like this:
```
parent/
parent/.denv/
parent/.denv/config
parent/subfolder/
parent/subfolder/ldmx-sw/
parent/subfolder/ldmx-sw/justfile
```
I'm cloning ldmx-sw into the "subfolder" directory, while a denv workspace already exists in a parent directory above it. 

Running `just init` would claim that the workspace is already initialized

```
ewallin@ludde:~/Programming/parent/subfolder/ldmx-sw$ just init
Workspace already initialized.
denv_workspace="/home/ewallin/Programming/parent"
denv_name="ldmx"
denv_image="ldmx/dev:latest"
[...]
```

This is because it finds a workspace higher up in the file hierarchy:
```
ewallin@ludde:~/Programming/parent/subfolder/ldmx-sw$ denv check --workspace
[...]
Found denv_workspace="/home/ewallin/Programming/parent"
```

But `just configure` only looks for the denv config file in `${LDMX_BASE}`, and thus fails:
```
ewallin@ludde:~/Programming/parent/subfolder/ldmx-sw$ just configure build
denv cmake -B build -S . -DADDITIONAL_WARNINGS=ON -DENABLE_CLANG_TIDY=ON build
/home/ewallin/.local/bin/denv: 488: .: cannot open /home/ewallin/Programming/parent/subfolder/.denv/config: No such file
error: Recipe `configure-base` failed on line 65 with exit code 2
```

tomeichlersmith, 2025-01-06:
In the short term, you can get around this by forcing `denv` to override the parent denv with the LDMX_BASE denv.

```
cd parent/subfolder
denv init ldmx/dev:latest
# respond "y" if asked about overriding (introduced in denv v1.1)
```

In the long term, we should decide if this is a workflow that should be supported. Having denvs within one another can easily lead to confusion since `cd ..` can change which denv is chosen to run. If we do want to support this workflow, then we would want to pass the prompts that `denv` issues to the user so they are at least acknowledging that they have nested denvs.

tvami, 2025-02-14:
This happened to one of my new student now

> In the long term, we should decide if this is a workflow that should be supported. Having denvs within one another can easily lead to confusion since cd .. can change which denv is chosen to run. If [...]

What I'd like is to have a warning that there is a `.denv` above, and the option to get rid of it.

This is what he saw
```
cd /home/henry/ldmx/noise_study/ldmx-sw/
just init 
/home/henry/.local/bin/denv: line 495: /home/henry/ldmx/noise_study/.denv/config: No such file or directory
error: Recipe `init` failed with exit code 1
```
while he had a `.denv` in
```
cd /home/henry/ldmx/.denv/
```

Deleting `/home/henry/ldmx/.denv/` made it work, but it would be good to streamline this for newcomers

tvami, 2025-02-14:
Btw, this is essentially a duplicate of https://github.com/LDMX-Software/ldmx-sw/issues/1472 
So closing it as dupplicate

closed 2025-02-14 by tvami