LDMX-Software/ldmx-sw #1513: stop having our detectors in vacuum

bryngemark, 2025-01-16:
**Is your feature request related to a problem? Please describe.**
We use vacuum in our gdmls. ESA will contain, among other pieces of junk and museum artifacts, air. 

**Describe the solution you'd like**
That we specify air as the material containing everything else. 

**Describe alternatives you've considered**
Leaving it as is. That is fine if we can convince ourselves it is. Which requires switching to air to compare. 

**Additional context**
Partly discussed already in issue #1435

tvami, 2025-01-16:
Can we do a trick where we define "vacuum" to have the density of air?

tvami, 2025-01-16:
Or maybe it's just that we replace 
https://github.com/LDMX-Software/ldmx-sw/blob/9690102c702d341b9d2fb208f7fdb8faf0be23fe/Detectors/data/ldmx-det-v14-8gev/detector.gdml#L65
with the already defined Air definition?

tvami, 2025-01-16:
I opened https://github.com/LDMX-Software/ldmx-sw/pull/1514 more of a curiosity of what it will do :D I can close it if it's not the way to go

closed 2025-01-20 by tvami