LDMX-Software/ldmx-sw #1520: Signal CI is not working now

tvami, 2025-01-18:
**Describe the bug**

The signal CI script takes the `latest` db-gen, which now uses the MG5 but the ldmx-sw still assumes MG4

**To Reproduce**

Just look at any recent CI results, they all show the somewhat useless msg of 
```
Unrecognized Exception: BadConf : Unable to find any library entries at 'electron_tungsten_MaxE_8.0_MinE_4.0_RelEStep_0.1_UndecayedAP_mA_0.01_run_1'
  The library is either a single CSV file or a directory of LHE files.
  Any individual file can be compressed with `gzip`.
  This means the valid extensions are '.lhe', '.lhe.gz', '.csv', and '.csv.gz'
error: Recipe `fire` failed on line 95 with exit code 127
```

Which actually makes me unhappy too: if it's BadConfig why is the CI passing??

**Desired behavior**

 * CI should have failed
 * We should not use `latest`, but update every time we think it makes sense to update

**Additional context**
Add any other context about the problem here.

tvami, 2025-01-18:
> We should not use `latest`, but update every time we think it makes sense to update 

--> I'll fix this

> CI should have failed 

--> I'm not sure what we need to change here, @tomeichlersmith ?

tomeichlersmith, 2025-01-19:
I am a bit surprised to see this. I thought the library parsing was relatively stable in G4DarkBreM (where this issue comes from).

I hope it is simply a directory crawling issue, but I'll look into it this coming week.

closed 2025-01-19 by tvami

tvami, 2025-01-19:
> I thought the library parsing was relatively stable in G4DarkBreM (where this issue comes from).

I dont think it's due to the library parsing, but the fact that the IDs of DM and electron and W have changed. And then the msg is just misleading. 

> CI should have failed

Do you have a guess why the CI didnt fail when ldmx-sw clearly had an exception?