LDMX-Software/ldmx-sw #1521: Change ldmx-sw to be able to read MG5 inputs

tvami, 2025-01-18:
**Is your feature request related to a problem? Please describe.**

We moved the LHE production to MG5 in https://github.com/LDMX-Software/dark-brem-lib-gen/pull/19

**Describe the solution you'd like**

Now next step is to change ldmx-sw to be able to read those LHE files

**Describe alternatives you've considered**

Keep using MG4 files

**Additional context**

Follow-up to https://github.com/LDMX-Software/ldmx-sw/issues/1520

tomeichlersmith, 2025-01-20:
I generated a MG5 library using v5.0 of dark-brem-lib-gen.
```
dbgen use latest
mkdir scratch
dbgen work scratch/
dbgen run --run 1 --max_energy 8.0 --apmass 0.01
```
The library generated has the correct structure. One directory with the LHE files within it.
The LHE files do have events in them. I am now guessing that the parsing library I wrote initially in G4DarkBreM does not support some additional LHE tags that are inserted by MG5.

Then I ran the config from the signal validation sample.
```
denv init ldmx/pro:v4.2.0
# copy ldmx-sw/.github/validation_samples/signal/config.py to this directory
# set p.maxEvents = 10 and p.run = 1
denv fire config.py
```
which replicates the issue from the CI.
```
[ ParallelWorld ] 0 debug: Adding : sp_hcal_front_back_PV to parallel world.
[ ParallelWorld ] 0 debug: Adding : sp_hcal_top_bot_PV to parallel world.
[ ParallelWorld ] 0 debug: Adding : sp_hcal_left_right_PV to parallel world.
Unrecognized Exception: BadConf : Unable to find any library entries at 'dblibgen/electron_tungsten_MaxE_8.0_MinE_4.0_RelEStep_0.1_UndecayedAP_mA_0.01_run_1'
  The library is either a single CSV file or a directory of LHE files.
  Any individual file can be compressed with `gzip`.
  This means the valid extensions are '.lhe', '.lhe.gz', '.csv', and '.csv.gz'
```

Luckily for me, past Tom figured that library parsing would be annoying and wrote the `g4db-extract-library` program which uses the LHE library parser from G4DarkBreM to extract the necessary components and then writes a CSV that is loadable by G4DarkBreM. Using this command runs successfully but produces an empty CSV.

tomeichlersmith, 2025-01-20:
I'm guessing the LHE parser, which made [some assumptions](https://ldmx-software.github.io/G4DarkBreM/namespaceg4db_1_1parse.html#ab215e493cc007d0ba5ec5d866e59e358) about the input LHE files, had those assumptions broken when we switched to MG5.

tomeichlersmith, 2025-01-20:
Found it. The new MG model uses the PDG ID 1023 for the dark photon but we assume the PDG ID here is 622. There are two solutions:
1. Update our code here to look for PDG 1023 in the LHE rather than PDG 622.
2. Update dark-brem-lib-gen to use 622 for the dark photon PDG.

I prefer option (1) since the PDG 1023 conforms with the PDG ID numbering standards and I know where to fix it.
And actually, the `aprime_lhe_id` is a configurable part of our dark brem model, so _technically_ resolving this issue doesn't require changing any of the C++! (Although, I think it would be a good idea to make the parameter default be the number produced by our default library generator.)

tomeichlersmith, 2025-01-20:
The nuclei IDs are ignored when parsing in G4DarkBreM. We just look for the target Z by scanning the file for the line with `Znuc` in it. I don't remember seeing a `611` milling around. 

Certainly, if we want our _Geant4_ PDG IDs to match the IDs in the LHE, then we need more C++ changes. Specifically, in how we define the `G4APrime`. The only place in ldmx-sw where we care about the nuclei ID is in the LHE Primary Generator.

https://github.com/LDMX-Software/ldmx-sw/blob/45ee57d8b344b4ef202d0b2cbbaa6afae5c49586/SimCore/src/SimCore/Generators/LHEPrimaryGenerator.cxx#L53-L57

But I think this fix is only necessary because the nuclei _was not_ conforming to the PDG ID standards. I think Geant4 will be able to deduce the correct nuclei from these IDs (untested).

PR #1525 (2025-01-24): update default LHE ID for A' to be 1023
This update then makes the default configuration of the dark brem model (from our point of view) align with the libraries that are produced by v5.0 of dark-brem-lib-gen.

Users can still use earlier versions of dark-brem-lib-gen, but then they will need to change their configuration script to look for the old LHE ID 622 rather than the new one.
```python
mySim.dark_brem.model.aprime_lhe_id = 622
```

This change *does not* alter the PDG ID of the dark photon within our simulation. It will still have the PDG ID of 622 according to Geant4 and the output Si [... cut]
closed 2025-01-24 by tomeichlersmith