LDMX-Software/ldmx-sw #1533: The `validate_detector` is broken

tvami, 2025-01-24:
**Describe the bug**

When adding 
```
mySim.validate_detector = True
```
to any configs, I get

```

---- LDMXSW: Loading configuration --------
Setting detector for tracking to ldmx-det-v14-8gev-no-cals
Setting detector for tracking to ldmx-det-v14-8gev
Aborted (core dumped)
error: Recipe `fire` failed on line 95 with exit code 134
```

**To Reproduce**
Steps to reproduce the behavior:

I'm in current master, i.e. `v4.2.3` 

Edit 
```
.github/validation_samples/ecal_pn/config.py
```
by adding
```
mySim.validate_detector = True
```
then
```
just setenv LDMX_NUM_EVENTS=1
just setenv LDMX_RUN_NUMBER=1
just fire .github/validation_samples/ecal_pn/config.py 
```

**Desired behavior**

I actually dont know how this looked like when it worked, but clearly not a segfault :)

**Environment:**
Output of `ldmx config`:
```
denv config print
denv_workspace="/home/vamitamas/iss1530-simcalorimeterhits-class-misses-a-few-getters"
denv_name="ldmx"
denv_image="ldmx/dev:latest"
denv_mounts=""
denv_shell="/bin/bash -i"
denv_network="true"
apptainer version 1.2.5
LDMX_NUM_EVENTS=10
LDMX_RUN_NUMBER=1
DISPLAY=localhost:11.0
DENV_NAME=ldmx
DENV_RUNNER=apptainer
DENV_IMAGE=ldmx/dev:latest
```

**Additional context**

I was originally trying on https://github.com/LDMX-Software/ldmx-sw/pull/1515

tomeichlersmith, 2025-01-24:
Is there anything in `G4cout.log` or `G4cerr.log`? Those are the files Geant4's messages get written to (we should probably redirect them to our logging system at some point).

https://github.com/LDMX-Software/ldmx-sw/blob/b447b66202b166d08b057300c5e5a13ddc5a6c5a/SimCore/include/SimCore/G4Session.h#L30-L31

I've seen the `Aborted (core dumped)` in the terminal with the Geant4 error printed in `G4cerr.log` before. Not sure if that's the case here.

tomeichlersmith, 2025-01-24:
Ope, and actually, `simulator` has its own `verbosity` flag which, if zero, the Geant4 messages are just thrown away.

https://github.com/LDMX-Software/ldmx-sw/blob/9458b6baf1a4d0beb04dba82073037b1e40cacc6/SimCore/src/SimCore/SimulatorBase.cxx#L151-L162

so

```python
mySim.verbosity = 5
```

is necessary as well.

(We will not be able to completely rid ourselves of the `verbosity` parameter because we will probably want to be able to inform Geant4 what verbosity we want, but we could remove this cutoff which I implemented before the logging system.)

tvami, 2025-01-24:
with the verbosity increased, this is what I see in the `G4cerr.log`

```
-------- EEEE ------- G4Exception-START -------- EEEE -------
*** G4Exception : InvalidRead
      issued by : G4GDMLRead::Read()
ERROR - Empty document or unable to validate schema!
        Check Internet connection is ON in case of schema
        validation enabled and location defined as URL in
        the GDML file - /home/vamitamas/iss1530-simcalorimeterhits-class-misses-a-few-getters/ldmx-sw/install/data/detectors/ldmx-det-v14-8gev/detector.gdml - being imported!
        Otherwise, verify GDML schema server is reachable!
*** Fatal Exception *** core dump ***
-------- EEEE -------- G4Exception-END --------- EEEE -------

*** G4Exception: Aborting execution ***
````

tvami, 2025-01-24:
Uh ok, so if I download the gdml.xsd and copy it into the `install` directory 

```
diff --git a/.github/validation_samples/ecal_pn/config.py b/.github/validation_samples/ecal_pn/config.py
index 79a256a..132c023 100644
--- a/.github/validation_samples/ecal_pn/config.py
+++ b/.github/validation_samples/ecal_pn/config.py
@@ -9,6 +9,8 @@ det = 'ldmx-det-v14-8gev'
 mySim = ecal.photo_nuclear(det,gen.single_8gev_e_upstream_tagger())
 mySim.beamSpotSmear = [20.,80.,0.]
 mySim.description = 'ECal PN Test Simulation'
+mySim.validate_detector = True
+mySim.verbosity = 5

 p.sequence = [ mySim ]

diff --git a/Detectors/data/ldmx-det-v14-8gev/detector.gdml b/Detectors/data/ldmx-det-v14-8gev/detector.gdml
index c52bb59..dbde249 100644
--- a/Detectors/data/ldmx-det-v14-8gev/detector.gdml
+++ b/Detectors/data/ldmx-det-v14-8gev/detector.gdml
@@ -3,7 +3,7 @@
 <!ENTITY constants SYSTEM "constants.gdml">
 <!ENTITY materials SYSTEM "materials.gdml">
 ]>
-<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd">
+<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="../../gdml.xsd">

   <define>

diff --git a/Detectors/data/ldmx-det-v14-8gev/ecal.gdml b/Detectors/data/ldmx-det-v14-8gev/ecal.gdml
index d20bf3b..44ea11b 100644
--- a/Detectors/data/ldmx-det-v14-8gev/ecal.gdml
+++ b/Detectors/data/ldmx-det-v14-8gev/ecal.gdml
@@ -2,7 +2,7 @@
 <!DOCTYPE gdml [
 <!ENTITY constants SYSTEM "constants.gdml">
 ]>
-<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd">
+<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="../../gdml.xsd">

   <!--
     The LDMX ECal Geant4 Simulation Model
diff --git a/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard5_assembly.gdml b/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard5_assembly.gdml
index f06730b..a80d436 100644
--- a/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard5_assembly.gdml
+++ b/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard5_assembly.gdml
@@ -1,6 +1,5 @@
 <?xml version="1.0" encoding="UTF-8" standalone="no" ?>
-<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd">
-
+<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="../../gdml.xsd">
   <define>
     <position name="nohole_motherboard5_assembly_v0" unit="mm" x="197.0682" y="8.166" z="380.9181"/>
     <position name="nohole_motherboard5_assembly_v1" unit="mm" x="97.69203" y="8.166" z="437.1063"/>
diff --git a/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard6_assembly.gdml b/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard6_assembly.gdml
index edde534..1826d2d 100644
--- a/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard6_assembly.gdml
+++ b/Detectors/data/ldmx-det-v14-8gev/ecal_motherboard6_assembly.gdml
@@ -1,5 +1,5 @@
 <?xml version="1.0" encoding="UTF-8" standalone="no" ?>
-<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd">
+<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="../../gdml.xsd">

  <define>
     <position name="nohole_motherboard6_assembly_v0" unit="mm" x="106.2316" y="8.166" z="10.69263"/>
diff --git a/Detectors/data/ldmx-det-v14-8gev/ecal_support_box.gdml b/Detectors/data/ldmx-det-v14-8gev/ecal_support_box.gdml
index c8a4d7c..d67593a 100644
--- a/Detectors/data/ldmx-det-v14-8gev/ecal_support_box.gdml
+++ b/Detectors/data/ldmx-det-v14-8gev/ecal_support_box.gdml
@@ -1,5 +1,5 @@
[... 84 lines cut ...]
 <!DOCTYPE gdml [
 <!ENTITY constants SYSTEM "constants.gdml">
 ]>
-<gdml xmlns:gdml="http://cern.ch/2001/Schemas/GDML" 
-    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
-    xsi:noNamespaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd" >
+<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="../../gdml.xsd">

   <define>

diff --git a/Detectors/data/ldmx-det-v14-8gev/trig_scint.gdml b/Detectors/data/ldmx-det-v14-8gev/trig_scint.gdml
index 62f0e23..9291693 100644
--- a/Detectors/data/ldmx-det-v14-8gev/trig_scint.gdml
+++ b/Detectors/data/ldmx-det-v14-8gev/trig_scint.gdml
@@ -2,9 +2,7 @@
 <!DOCTYPE gdml [
 <!ENTITY constants SYSTEM "constants.gdml">
 ]>
-<gdml xmlns:gdml="http://cern.ch/2001/Schemas/GDML" 
-    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
-    xsi:noNamespaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd" >
+<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="../../gdml.xsd">

   <define>
```

tvami, 2025-01-24:
`G4cerr` is empty, but `G4cout` has a lot of validation warnings, like
```
G4GDML: VALIDATION ERROR! type ':FileReferenceType' not found at line: 22
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 23
G4GDML: VALIDATION ERROR! type ':positionType' not found at line: 26
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 27
G4GDML: VALIDATION ERROR! type ':rotationType' not found at line: 30
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 31
G4GDML: VALIDATION ERROR! type ':scaleType' not found at line: 34
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 35
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 48
G4GDML: VALIDATION ERROR! simpleType ':ExpressionOrIDREFType' for attribute 'number' not found at line: 51
G4GDML: VALIDATION ERROR! simpleType ':ExpressionOrIDREFType' for attribute 'width' not found at line: 52
G4GDML: VALIDATION ERROR! simpleType ':ExpressionOrIDREFType' for attribute 'offset' not found at line: 53
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 67
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 68
G4GDML: VALIDATION ERROR! type ':replicavol' not found at line: 72
G4GDML: VALIDATION ERROR! referenced element 'replicavol' not found at line: 72
G4GDML: VALIDATION ERROR! type ':ParameterisedPlacementType' not found at line: 73
G4GDML: VALIDATION ERROR! type ':loop' not found at line: 75
G4GDML: VALIDATION ERROR! referenced element 'loop' not found at line: 75
G4GDML: VALIDATION ERROR! type ':AuxiliaryType' not found at line: 76
G4GDML: VALIDATION ERROR! type ':replicavol' not found at line: 96
G4GDML: VALIDATION ERROR! referenced element 'replicavol' not found at line: 96
G4GDML: VALIDATION ERROR! type ':ParameterisedPlacementType' not found at line: 97
G4GDML: VALIDATION ERROR! simpleType ':ExpressionOrIDREFType' for attribute 'surfaceproperty' not found at line: 109
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 126
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 127
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 142
G4GDML: VALIDATION ERROR! type ':loop' not found at line: 158
G4GDML: VALIDATION ERROR! referenced element 'loop' not found at line: 158
G4GDML: VALIDATION ERROR! type ':ParameterisationAlgorithm' not found at line: 159
G4GDML: VALIDATION ERROR! referenced element 'ParameterisationAlgorithm' not found at line: 159
G4GDML: VALIDATION ERROR! type ':define' not found at line: 169
G4GDML: VALIDATION ERROR! referenced element 'define' not found at line: 169
G4GDML: VALIDATION ERROR! type ':materials' not found at line: 170
G4GDML: VALIDATION ERROR! referenced element 'materials' not found at line: 170
G4GDML: VALIDATION ERROR! type ':solids' not found at line: 171
G4GDML: VALIDATION ERROR! referenced element 'solids' not found at line: 171
G4GDML: VALIDATION ERROR! type ':AuxiliaryType' not found at line: 176
G4GDML: VALIDATION ERROR! type ':ReferenceType' not found at line: 188
G4GDML: VALIDATION ERROR! no declaration found for element 'define' at line: 4
G4GDML: VALIDATION ERROR! no declaration found for element 'solids' at line: 496
G4GDML: VALIDATION ERROR! element 'define' is not allowed for content model '(structure,userinfo?,setup+)' at line: 1491
```

tomeichlersmith, 2025-01-24:
aw geez, we've been forgetting to validate 😬

tomeichlersmith, 2025-01-24:
I was able to fetch the schema from within the container manually, so I'm hoping that part was just a fluke. BUT we are using a wildly old version of Geant4 so I also wouldn't be surprised if its fetching code is not working anymore.

```
tom@appa:~/code/ldmx/ldmx-sw$ denv wget http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd
--2025-01-24 20:49:27--  http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd
Resolving service-spi.web.cern.ch (service-spi.web.cern.ch)... 2001:1458:d00:65::100:265, 2001:1458:d00:16::41d, 2001:1458:d00:62::100:2e8, ...
Connecting to service-spi.web.cern.ch (service-spi.web.cern.ch)|2001:1458:d00:65::100:265|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd [following]
--2025-01-24 20:49:28--  https://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd
Connecting to service-spi.web.cern.ch (service-spi.web.cern.ch)|2001:1458:d00:65::100:265|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10138 (9.9K) [text/xml]
Saving to: 'gdml.xsd'

gdml.xsd                      100%[===============================================>]   9.90K  --.-KB/s    in 0s      

2025-01-24 20:49:28 (266 MB/s) - 'gdml.xsd' saved [10138/10138]
```

tvami, 2025-01-24:
Yes, the wget works, but when I try to let GEANT access it, it somehow doesnt access it anymore. Did that part work for you?

tomeichlersmith, 2025-01-24:
Nope, same error as you :/ crap

tvami, 2025-07-14:
It's not a real solution, but while working on
https://github.com/LDMX-Software/ldmx-sw/pull/1767

I realized that by commenting out
https://github.com/LDMX-Software/ldmx-sw/blob/trunk/SimCore/src/SimCore/SimulatorBase.cxx#L152
```
// sessionHandle_ = std::make_unique<BatchSession>();
```
I get the verbose, detector validation msgs that I believe was the intention of this `validate_detector` flag. (But maybe there is more; nevertheless I think this is still progress)