LDMX-Software/ldmx-sw #1538: ldmx-sw version of previously generated files

jpasc27, 2025-01-27:
**Is your feature request related to a problem? Please describe.**
I want to ensure samples that I am generating are using the same *ldmx-sw* version, but there is no way for me to check what version I used in my previously generated files, as far as I know.

**Describe the solution you'd like**
It would be good if there was a simple command to do this, but at the very least, we need the *ldmx-sw* version encoded into the generated files in some readable manner.

**Describe alternatives you've considered**
Writing down (externally) the version number for every generated file is a possibility, but tedious, and information can easily slip through the cracks as files are transferred, re-run, etc. 

**Additional context**
@tvami and @tomeichlersmith had a conversation on Slack about this issue, perhaps they can summarize what was discussed, or continue the conversation here.

tvami, 2025-01-27:
We discussed a path forward: that is we will keep the version number in the run header, and we discussed the technical implementation which I dont think is worth repeating it here, but rather do it in the PR.

tomeichlersmith, 2025-01-27:
General idea is to read from `.github/actions/validate/gold_label` so that the last version is stored in the RunHeader.
 
1. `file(READ ...)` in CMake
2. `configure_file` to put the version string into Framework/include/Framework/Verison.h
3. RunHeader includes Version.h so it can store the values there as member variables

This process is already done for `GIT_SHA` but the command that CMake runs to deduce GIT_SHA (`git rev-parse HEAD`) fails within the container because the container does not have `git` installed.
We could install `git` into the container, but I don't want to do that because git's ID-necessary processes do not function from within the container on all systems. https://github.com/tomeichlersmith/denv/issues/102
Instead, I think we could use `file(READ ...)` in CMake and parse the GIT_SHA from the `.git` tree. I've done this before and it works pretty well.

https://github.com/LDMX-Software/fire/blob/b65921f52f4e1f669a9e6fba3d983d8de0f86412/CMakeLists.txt#L39-L61