LDMX-Software/ldmx-sw #1552: CI needs further fixing

tvami, 2025-01-31:
**Describe the bug**

The env variable is deduced to be
```
  LDMX_VERSION_FILE=/home/runner/work/ldmx-sw/ldmx-sw/VERSION
```

**To Reproduce**

For example see
https://github.com/LDMX-Software/ldmx-sw/actions/runs/13061170684/job/36445294614

**Desired behavior**

Move it one directory up

tvami, 2025-01-31:
Maybe the issue is that `GITHUB_WORKSPACE` changes for the different CIs

tvami, 2025-01-31:
Thinking more, I think this will break the gold generation CI, otherwise this
https://github.com/LDMX-Software/ldmx-sw/pull/1550/files
would have failed

tomeichlersmith, 2025-01-31:
The problem is not with the placement of `VERSION`, the problem is how we distribute the gold to the various validation jobs.

We copy the gold over from trunk, but then only tar-ball up the install and the `.github` directory. We need to include `VERSION` in the tar that is then shipped to the other jobs.

https://github.com/LDMX-Software/ldmx-sw/blob/5d9db0f92082d17c37d00631397efa4b0fad7861/.github/workflows/pr_validation.yml#L37-L42

PR #1554 (2025-01-31): include VERSION in ldmx-sw package for PR validation
the VERSION file is shipped with the gold that it originates from so that we can deduce a label for the reference histograms in the plots.

I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1552 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.
closed 2025-01-31 by tomeichlersmith