LDMX-Software/ldmx-sw #1562: MagneticFieldError with propagation interpolation in signal

tvami, 2025-02-02:
**Describe the bug**

```
[ ecalVeto ] 122  info:  Recoil tracking for projections; with 1 tracks, fiducial in tracker = true
  [ ecalVeto ] 122  info: Finding linreg tracks from a total of 6 hits using a radius of 35 mm
  [ ecalVeto ] 122  info:  MIP tracking completed; found 0 straight tracks and 0 lin-reg tracks
  [ ecalVeto ] 122  info:   BDT was ran, score is 0.999697
  [ hcalVeto ] 122  info: There are 0 / 0 HCal hits read out. 0 are not associated with the recoil ele. Total PE of 0
  [ hcalVeto ] 122  info: HCAL veto passed? true
  [ ElectronCounter ] 122  info: Found 1 electrons (tracks) using input collection TriggerPadTracksY_
  [ trigger ] 122  info: Got trigger energy cut 3000 for 1 electrons counted in the event.
  [ trigger ] 122  info: Got trigger energy sum 807.849; and decision is pass = true
  05:29:18    PROP           ERROR     Step failed with MagneticFieldError:1: Interpolation out of bounds was requested
  
   *** Break *** segmentation violation
```

**To Reproduce**
Add tracking in the signal CI, the inverse of this commit: https://github.com/LDMX-Software/ldmx-sw/pull/1551/commits/3c61e7093d429fa16270cc4b88111c90ed489360

**Desired behavior**

Not have a seg fault...

**Environment:**

This happened in the github CI env, with ldmx-sw 4.2.8

**Additional context**

Came up here
https://github.com/LDMX-Software/ldmx-sw/pull/1551#issuecomment-2629112746

tvami, 2025-02-05:
This comes up in bkg too, although no seg fault, here is the huge output from ACTS

```
[ Recoil_TrackFinder ] 5  info:     Could not extrapolate to ECAL!! Please check the track states
Surface::93.5
 -48
 -40
Smoothed:: -22.2343   38.0839 -0.597976   1.54822  -16.4095   99.7591
HasSmoothed::1
Filtered:: -22.2343   38.0839 -0.597976   1.54822  -16.4095   99.7591
21:03:59    ACTS_PROP      VERBOSE   Path aborter limit set to 658.7 (full helix = 1317.4, previous limit = 1.79769e+308)
21:03:59    ACTS_PROP      VERBOSE   Entering propagation.
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material on surface vol=3|lay=10|sen=9
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | x0=95.7|l0=465.2|ar=28.03|z=14|molar_rho=8.27685e-05|t=0.387284 absPdg=e- mass=0.510999MeV momentum=0.0609402GeV energyloss=0.164588MeV
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 177.909
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to ( +∞,  +∞, 177.909,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to ( +∞,  +∞, 177.909,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Starting stepping loop.
21:03:59    ACTS_PROP      VERBOSE   Step with size = 1.45232 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 177.3
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to ( +∞, 0.00345436, 177.3,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to ( +∞, 0.00345436, 177.3,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Step with size = 0.00345436 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 177.298
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to ( +∞,  +∞, 177.298,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to ( +∞,  +∞, 177.298,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Step with size = 1.61733 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 176.623
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to ( +∞, 0.00431707, 176.623,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to ( +∞, 0.00431707, 176.623,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Step with size = 0.00431707 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 176.622
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to ( +∞,  +∞, 176.622,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to ( +∞,  +∞, 176.622,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Step with size = 6.67359 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 173.873
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to (5.32558, 102.447, 173.873,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to (5.32558, 102.447, 173.873,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Step with size = 5.32558 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 171.713
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to (5.40895, 98.9663, 171.713,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to (5.40895, 98.9663, 171.713,  +∞)
21:03:59    ACTS_PROP      VERBOSE   Step with size = 5.40895 performed
21:03:59    ACTS_PROP      VERBOSE   MaterialInteractor | Found material in volume vol=3
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 169.545
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to (5.52258, 95.403, 169.545,  +∞)
[... cut ...]
21:03:59    ACTS_PROP      VERBOSE    -> near limit, far limit, distance: -0.1, 1.79769e+308, 57.6369
21:03:59    ACTS_PROP      VERBOSE   Intersection is WITHIN limit
21:03:59    ACTS_PROP      VERBOSE   SurfaceReached aborter | Target stepSize (surface) updated to (9.34768, 57.6369, 57.6369,  +∞)
21:03:59    ACTS_PROP      VERBOSE   PathLimit aborter | Target stepSize (path limit) updated to (9.34768, 57.6369, 57.6369,  +∞)
21:03:59    ACTS_PROP      ERROR     Step failed with MagneticFieldError:1: Interpolation out of bounds was requested
Propagation failed
```

tvami, 2025-02-05:
Looking at the magnetic field map, it does seem to end at 1500 mm in Z so that should cover both the ECAL and the side-HCAL.

But then it seems in Y we only go from -70 mm to 70 mm, which I dont understand since I'd assume that we want to cover the ECAL XY face (having X at 250 makes sense, I'd would have thought Y goes up to the same number).

```
-250.0 -55.0 1490.0 2.029E-07 -4.495E-06 -8.496E-07 
-250.0 -55.0 1495.0 1.980E-07 -4.427E-06 -8.140E-07 
-250.0 -55.0 1500.0 1.931E-07 -4.359E-06 -7.783E-07 
-250.0 -50.0 -1500.0 1.758E-07 -4.369E-06 7.085E-07 
-250.0 -50.0 -1495.0 1.803E-07 -4.438E-06 7.410E-07 
-250.0 -50.0 -1490.0 1.847E-07 -4.507E-06 7.735E-07 
```

@tomeichlersmith is there a way to easily add zeros everywhere else? At one point I remember discussions about having this in the Conditions, would that help this?

tomeichlersmith, 2025-02-05:
The reason we haven't run into this before is because the simulation just sets the Bfield outside of the interpolatable region to zero

https://github.com/LDMX-Software/ldmx-sw/blob/ea2a21c94057cf6eba09ce2b75f555976f6097df/SimCore/src/SimCore/MagneticFieldMap3D.cxx#L253-L257

We could potentially put this kind of object in the conditions and then both the simulation and tracking would access it; however, I am unsure on how tracking accesses the magnetic field. If we can provide a simple specialization (like how we have a concrete derived class of `G4MagneticField`), then I think we could make it work well.

bloodyyugo, 2025-02-05:
Here is where CKFProcessor gets the field and gives it to ACTS (via the Stepper). 

https://github.com/LDMX-Software/ldmx-sw/blob/ea2a21c94057cf6eba09ce2b75f555976f6097df/Tracking/src/Tracking/Reco/CKFProcessor.cxx#L118

...loads it in from the text file.  I'm a bit surprised the stepper doesn't try to do any extrapolation outside of the given limits.  Let me look if there are any options for that (or maybe just assume 0 field instead of just crapping out).

tomeichlersmith, 2025-02-05:
ah yes, I'm remembering working with PF now. It needs to be rotated into tracking coordinates so it can't be the exact same function as the sim.

Do we control `InterpolatedMagneticField3` or is it something we could override ourselves?

bloodyyugo, 2025-02-05:
The InterpolatedMagneticField3 is defined here:  
https://github.com/LDMX-Software/ldmx-sw/blob/ea2a21c94057cf6eba09ce2b75f555976f6097df/Tracking/include/Tracking/Sim/BFieldXYZUtils.h#L18

`using InterpolatedMagneticField3 = Acts::InterpolatedBFieldMap<
    Acts::Grid<Acts::Vector3, Acts::Axis<Acts::AxisType::Equidistant>,
               Acts::Axis<Acts::AxisType::Equidistant>,
               Acts::Axis<Acts::AxisType::Equidistant>>>;`

So, we sort of control it, but not the base "InterpolatedBFieldMap".  One thing we might be able to do is to add a bin with 0 field at large +/- y (we'd have to use the variable bin size axis for this, but that's supported).  

Or, of course we can just make the map bigger and add lots of 0-field entries (or whatever, do our own exterpolation).  Dumb, but easy.

tvami, 2025-02-05:
> Or, of course we can just make the map bigger and add lots of 0-field entries (or whatever, do our own exterpolation). Dumb, but easy.

This might be the quick-and-dirty solution for short term, and we can think about something better for the long term.
@tomeichlersmith do you have the code that produced this txt file? Or this is something we just took from HPS as is?

tomeichlersmith, 2025-02-05:
This was something that we took from HPS and then "expanded" to fit the space of the LDMX dipole. I inherited from my predecessors long ago and so the code doing this expansion has been lost to time.

tvami, 2025-02-21:
So I made new map where I add a lot of zeros... but the file gets huge, and I even notice the load time in runtime when I use it.

I plotted the values for each dimension, and each B field dimension (so 9 plots in total)
A few things tho that I see and dont understand. Well let's start with what I do:

[image]

[image]

[image]

Blue is original, red is my addition, makes sense for both.

[image]

[image]

[image]

Bz vs X (also By vs X) transition could be better, but at least it's going towards zero at 240.

The Bx vs X tho didnt change sign yet in the original map, so this doesnt look great...

And now it will not become better:

[image]

[image]

[image]

@bloodyyugo @tomeichlersmith thoughts?

tomeichlersmith, 2025-02-21:
I have never looked at the field so I'm not sure if these kinds of discontinuities make sense.

tvami, 2025-02-22:
The 3D maps:
[image]
[image]
[image]
[image]

tomeichlersmith, 2025-02-24:
Okay, that makes sense to me, the Bx and Bz are mainly inside the big hunks of metal and then they drop off once outside?

closed 2026-02-05 by tvami