LDMX-Software/ldmx-sw #1570: Ability to run on a run range

tvami, 2025-02-04:
**Is your feature request related to a problem? Please describe.**

I'd like to have the ability to run on a run range with there is an input file.
This I'd need now to debug events, I dont need to resimulate, just run the tracking on the n-th to m-th event.

I also think that this is a feature we'll need when we have real data, you often want to debug some weird event at event number 124241

**Describe the solution you'd like**

I think here
https://github.com/LDMX-Software/ldmx-sw/blob/ea2a21c94057cf6eba09ce2b75f555976f6097df/Framework/src/Framework/Process.cxx#L333
this needs to become a for loop and we add the lower and upper ends in there.

**Describe alternatives you've considered**

Even to resimulation should work on simulation, that's not a solution to real data.

**Additional context**

For example in https://github.com/LDMX-Software/ldmx-sw/issues/1562 happens at 122nd event

tomeichlersmith, 2025-02-04:
The biggest annoyance is skipping events efficiently *and correctly*. `nextEvent` does a lot of things and I am unsure if we just update the event number if it will behave properly.

tvami, 2025-02-04:
>  we could scan the input file manually in order to reach the correct starting event.

what do you mean on this?

tomeichlersmith, 2025-02-04:
something like
```cpp
while(n_events_processed < starting_event_index && masterFile->nextEvent()) {
  n_events_processed++;
}
```
instead of trying to set the index and skip to the right event. (TFile thinks we are reading these events but we are just not giving them to the processor sequence.)

closed 2025-02-10 by tvami