LDMX-Software/ldmx-sw #1573: Gold log files are always one version behind

tvami, 2025-02-05:
**Is your feature request related to a problem? Please describe.**

When a new gold is added
https://github.com/LDMX-Software/ldmx-sw/pull/1567/files#diff-7b60b8e351cbb80c47459ffe2c79f1a26404871f49294780fe47ad0e58c09350

We update VERSION to be the version that is the patch we are updating to. But the logs still contain the version before the update. With this the test are perpetually failing on the log diff.

[image]

**Describe the solution you'd like**

Have these the same, tho I don't see what we did wrong in the first place.

**Describe alternatives you've considered**

Just not be annoyed that the checks are never perfect

tvami, 2025-02-05:
> Have these the same, tho I don't see what we did wrong in the first place.

@tomeichlersmith do you see a solution to this?

tomeichlersmith, 2025-02-05:
We'd have to update VERSION and then build and run ldmx-sw. Right now, it is building and running ldmx-sw and then making one update of the gold and VERSION, but that means the build constructing the log uses the previous VERSION.

closed 2025-02-06 by tvami