LDMX-Software/ldmx-sw #1579: Fix label in PR validation comparison plots

tvami, 2025-02-06:
**Describe the bug**

Now if we trigger tests by comment, it runs on the correct branch but doesnt include the correct label in the plot
https://github.com/LDMX-Software/ldmx-sw/pull/1574#issuecomment-2638881997

tvami, 2025-02-07:
@tomeichlersmith it seems https://github.com/LDMX-Software/ldmx-sw/pull/1580 didnt work:

I look at HCAL 
[image]

The red is the new gold, but the blue is called `trunk`, even tho it's not actually `trunk`:

https://github.com/LDMX-Software/ldmx-sw/actions/runs/13191941735/job/36826359371

had
```
Switched to a new branch 'iss1570-run-on-an-event-range'
```

[image]

tvami, 2025-02-07:
What's making it even more confusing is that the gold diff shows

```
100319,100320c100319,100320
  <     ldmx-sw revision = 9bf6acc1544ed7e7ac2834eb5d0a16cccefa6499
  <     ldmx-sw version = 4.2.8
  ---
  >     ldmx-sw revision = 6edb29d9fdf546707809155e3d7
  >     ldmx-sw version = v4.2.9
  100327c100327
```

how can this be `v4.2.9` if the plot has `v4.2.10`?

tvami, 2025-02-07:
There is two issues...

1) When I trigger tests with a comment it seems to take the branch as it is at the time when it was created, this this would only work if somebody rebased... Not for this issue...

2) When I do it with the "ready for review" 
```
Compare to Golden Histograms
  Traceback (most recent call last):
    File "/home/runner/work/ldmx-sw/ldmx-sw/./.github/actions/validate/compare.py", line 195, in <module>
      compare(sys.argv[1], sys.argv[2], sys.argv[3], sys.argv[4])
  IndexError: list index out of range
```
https://github.com/LDMX-Software/ldmx-sw/actions/runs/13193852544/job/36832068570?pr=1578

which suggests `${HEAD_REF}` was empty at the `validate.sh` level :(

tomeichlersmith, 2025-02-07:
My next guess at this time is that the environment variables from one job are not persisted to the other jobs meaning the job step that deduces HEAD_REF needs to be done again at the beginning of all the `generation` jobs.

https://github.com/LDMX-Software/ldmx-sw/blob/cf851e27dbae387e7d031f01f0ea1b305f6dd017/.github/workflows/pr_validation.yml#L38-L45

tvami, 2025-02-07:
Still not working :( 

https://github.com/LDMX-Software/ldmx-sw/actions/runs/13203022477/job/36860548762?pr=1578

tvami, 2025-02-12:
This:
https://github.com/LDMX-Software/ldmx-sw/commit/a868e9d29c2a690924d1913bd9445a0ec0ff67f3

made it work for both! Closing this issue now as completed

closed 2025-02-12 by tvami