LDMX-Software/ldmx-sw #1586: Negative energy hit in ecal PN

tvami, 2025-02-10:
**Describe the bug**

https://github.com/LDMX-Software/ldmx-sw/actions/runs/13248358110/job/36981301291?pr=1585
```
> [ ecalVeto ] 7403 fatal: ECal hit has negative or zero energy, this should never happen, check the threshold settings in HgcrocEmulator
```

**To Reproduce**
Run the ecalPN CI with seed = 1 and it will show up at 7403th event

**Desired behavior**
A clear and concise description of what you want to happen (i.e. instead of the bug).

**Environment:**

```
 ldmx-sw revision = 53c73d52a6ffee68f125cb31b378a66335a5b5b7
```

**Additional context**

Came up in https://github.com/LDMX-Software/ldmx-sw/pull/1585

tvami, 2025-02-11:
This is the RecHit info
```
EcalHit { id: 140200a3, energy: 0MeV, time: 0ns, amplitude: 0}
```
the SimHit info
```
SimCalorimeterHit { id: 335675555,  edep: 5.81587, position: ( -13.1901, 6.78159, 255.969 ), num contribs: 1 }
```
where the contrib PDG ID is -11

Now looking at the digitization, this printout
https://github.com/LDMX-Software/ldmx-sw/blob/fac6e765712aa506afc3045a6ea70b2e464b7454/Ecal/src/Ecal/EcalDigiProducer.cxx#L138-L142
give
```
335675555 5.81587 MeV at 78.0936
```
so the pulse is at 78.09 ns (is that the unit?). There are 5 more simhits with 78 ns, but those dont produce null energy rechits. The rests are at 0.1-0.01 level

tomeichlersmith, 2025-02-11:
The time units are ns, those are very late hits (For context, we expect to calibrate the detector so an in-time pulse happens at 0ns).

Is this just the lowest energy, super late hit? If the others are high enough energy, maybe they don't get rounded to zero. The emulation's check on above/below threshold is _not_ dependent on the time (which may be a bug, should we require the hits to be above the readout threshold in the SOI?)

tvami, 2025-02-11:
This is the full list of hits below.
I see for example `335675535 1.65327 MeV at 78.1254` so certainly not the lowest energy, latest hit
```

335544480 0.0947531 MeV at 0.000710554
335544500 0.2696 MeV at 0.21917
335544596 0.286864 MeV at 0.868373
335675480 1.21645 MeV at 78.242
335675535 1.65327 MeV at 78.1254
335675552 0.114383 MeV at 0.000714771
335675554 0.979658 MeV at 78.039
335675555 5.81587 MeV at 78.0936
335675556 0.449145 MeV at 78.0981
335675572 0.241323 MeV at 0.131066
335806474 0.0964706 MeV at 0.315285
335806601 0.183776 MeV at 0.0577147
335806624 0.400652 MeV at 0.000722924
335806646 0.0838591 MeV at 0.00266693
335806647 0.770029 MeV at 0.00150953
335806695 0.121645 MeV at 0.716063
335827353 0.562816 MeV at 0.939308
335937695 0.382684 MeV at 0.00485073
335937696 0.810226 MeV at 0.000756607
335937697 1.11947 MeV at 0.00228213
335937719 2.75392 MeV at 0.00248246
335937766 0.000104869 MeV at 37.5987
335937790 0.870044 MeV at 0.623786
336068767 0.948327 MeV at 0.00855461
336068768 1.1473 MeV at 5.23656e-05
336068790 0.335474 MeV at 0.000539455
336068791 0.262722 MeV at 0.000612649
336068792 1.78893 MeV at 0.00404528
336068793 0.361322 MeV at 0.0267154
336068816 0.0444546 MeV at 0.0280082
336068839 0.129259 MeV at 0.086054
336068862 0.144342 MeV at 0.515698
336199761 0.107886 MeV at 0.10959
336199795 0.126281 MeV at 0.171168
336199800 0.199509 MeV at 0.0188825
336199817 0.0728034 MeV at 0.0109095
336199818 0.444825 MeV at 0.00304977
336199820 0.0111444 MeV at 0.0134683
336199840 0.901585 MeV at 0.00055914
336199841 1.26088 MeV at 0.00446143
336199842 0.202245 MeV at 0.010429
336199862 0.298993 MeV at 0.00150719
336199864 0.150274 MeV at 0.023045
336199886 0.51593 MeV at 0.0293859
336330833 0.173254 MeV at 0.104179
336330888 0.163963 MeV at 0.0149012
336330894 0.102056 MeV at 0.0510467
336330912 0.445346 MeV at 0.000969639
336330913 0.513376 MeV at 0.00774824
336330914 1.07044 MeV at 0.0072999
336330915 0.216789 MeV at 0.0195901
336330936 2.33851 MeV at 0.00322198
336330937 0.403712 MeV at 0.00754385
336330960 0.098616 MeV at 0.014959
336331004 0.0869261 MeV at 0.303254
336331026 0.297802 MeV at 0.305837
336461921 0.0965843 MeV at 0.023889
336461923 0.154018 MeV at 0.132301
336461924 0.168309 MeV at 0.0266527
336461940 0.0937746 MeV at 0.0173269
336461942 0.25538 MeV at 0.00944352
336461962 1.03209 MeV at 0.00585361
336461963 0.115449 MeV at 0.00384982
336461966 0.127517 MeV at 0.0206499
336461983 0.0417968 MeV at 0.0758165
336461984 0.681833 MeV at 0.00032397
336461986 1.74315 MeV at 0.00706549
336461987 0.0821861 MeV at 0.0172774
336462006 0.205136 MeV at 0.000874956
336462009 0.0830642 MeV at 0.0124711
336462010 1.27661 MeV at 0.0291652
336462053 0.355931 MeV at 0.192963
336592955 0.261988 MeV at 0.0897428
336592975 0.110633 MeV at 0.0365418
336592999 0.25465 MeV at 0.0855405
336593011 0.152543 MeV at 0.0379031
336593013 0.292483 MeV at 0.0134432
336593033 0.172953 MeV at 0.0559325
336593034 0.130268 MeV at 0.0142208
336593037 0.225816 MeV at 0.0447571
336593039 0.132582 MeV at 0.0516945
336593042 0.13149 MeV at 0.085956
336593055 0.138209 MeV at 0.00591974
336593056 0.293256 MeV at 0.00194614
336593057 0.143286 MeV at 0.00851064
336593060 0.378968 MeV at 0.0241819
336593061 0.433355 MeV at 0.0301617
336593078 0.428578 MeV at 0.0013303
336593079 0.850552 MeV at 0.0191782
336593081 0.392316 MeV at 0.0249867
336593082 0.0335515 MeV at 0.0654062
336593084 0.117267 MeV at 0.0335468
336593101 0.595678 MeV at 0.146972
336593103 0.0432174 MeV at 0.0558867
336593107 0.0022435 MeV at 0.0858337
336605243 0.986691 MeV at 0.516741
336724063 0.151888 MeV at 0.0615826
336724092 0.213488 MeV at 0.248141
336724104 0.243953 MeV at 0.0550753
336724107 0.114843 MeV at 0.00156144
336724108 1.64154 MeV at 0.0121323
336724113 0.137079 MeV at 0.0584035
336724125 0.245194 MeV at 0.0968084
336724126 0.254116 MeV at 0.0691892
336724128 0.145207 MeV at 0.00337175
336724129 0.315823 MeV at 0.00239853
336724134 0.550036 MeV at 0.0627786
336724148 0.761577 MeV at 0.101754
336724151 2.27774 MeV at 0.0154655
336724244 1.28933 MeV at 0.122514
336744599 0.472043 MeV at 0.220915
336744621 0.227429 MeV at 0.211236
336855112 0.498272 MeV at 0.304354
336855153 0.170216 MeV at 0.101359
336855159 0.033673 MeV at 0.0245786
336855180 0.0962346 MeV at 0.00705337
336855186 0.00738475 MeV at 0.1537
336855201 0.548856 MeV at 0.00299942
336855204 0.176599 MeV at 0.0165933
336855226 0.430343 MeV at 0.0192071
336855248 0.251055 MeV at 0.0148702
336855329 0.893266 MeV at 0.103565
336871646 0.276402 MeV at 0.41731
336986208 0.0757478 MeV at 0.298335
336986231 0.602141 MeV at 0.0115981
336986255 0.126851 MeV at 0.0354231
336986352 0.224827 MeV at 0.419013
337006953 0.205032 MeV at 0.135452
337117253 0.153683 MeV at 0.0579158
337117329 0.1958 MeV at 0.0369448
337117344 0.355783 MeV at 0.00172954
337117345 0.0928467 MeV at 0.00642186
337117347 0.149086 MeV at 0.0216909
337117348 0.286472 MeV at 0.0127734
337117366 0.885302 MeV at 0.0186908
337117373 0.140573 MeV at 0.0694832
337117389 0.100793 MeV at 0.00210565
337121285 0.294124 MeV at 10263
337138024 0.244861 MeV at 0.194452
337248380 0.130761 MeV at 0.0360956
337248395 0.252483 MeV at 0.00200334
337248420 0.0873079 MeV at 0.0122104
337269078 0.211684 MeV at 0.229138
337379402 0.179045 MeV at 0.11367
337379469 0.143173 MeV at 0.0197518
337379475 0.201981 MeV at 0.048517
337379562 0.333253 MeV at 0.0398228
337400148 0.246437 MeV at 0.291612
337510538 0.0852281 MeV at 0.00278638
337510555 0.18162 MeV at 0.0199582
337531180 0.1196 MeV at 0.332697
337531200 1.25491 MeV at 0.417382
337531267 0.300054 MeV at 0.133873
337772608 0.0342293 MeV at 0.0298131
337772625 0.763087 MeV at 0.0191702
337911838 0.490489 MeV at 0.176507
338297114 0.0270358 MeV at 0.119876
338575738 0.229199 MeV at 0.132877
```

tomeichlersmith, 2025-02-11:
Ok, I think we need to be applying the thresholds in time. I think what could be happening is that this hit is not vetoed by the readout threshold since it is high enough energy to go into TOT mode. Then since it is so late in time, the TOT mode is not triggered until several digi samples later which are then reconstructed as zero since the SOI is a zero sample. The other late-time hits would be vetoed by the readout threshold which is applied in the SOI.

TOT Section of Emulation

https://github.com/LDMX-Software/ldmx-sw/blob/fac6e765712aa506afc3045a6ea70b2e464b7454/Tools/src/Tools/HgcrocEmulator.cxx#L167-L182

ADC Section of Emulation

https://github.com/LDMX-Software/ldmx-sw/blob/fac6e765712aa506afc3045a6ea70b2e464b7454/Tools/src/Tools/HgcrocEmulator.cxx#L216-L218

I'm not sure what the "correct" behavior in this case is. Would we get these very-late digis from the real chip? If so, we should patch the reconstruction to be able to re-veto any digis below the readout threshold/pedestal. If the chip should never give us these digis, then we should patch the emulation with some check in TOT mode.

tvami, 2025-02-11:
let me tag @jmmans

jmmans, 2025-02-12:
Looks like this is a very late hit, right?  It should be zero for the current crossing...

tvami, 2025-02-12:
> Looks like this is a very late hit, right?

Yes, very late! 

>  It should be zero for the current crossing...

You mean that the chip will not read it out? If that's the case, where is the limit in the timing for the chip? 

Btw, would it read it out in the next crossing?

tvami, 2025-02-13:
I made printouts for the hits that are very late
```
[ ecalDigis ] 7403 debug: 337121285 0.294124 MeV at 10263
[ HgcrocEmulator ] 7403 debug:   iADC = 0 at startBX = -50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 1 at startBX = -25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 2 at startBX = 0
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 3 at startBX = 25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 4 at startBX = 50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 5 at startBX = 75
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 6 at startBX = 100
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 7 at startBX = 125
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 8 at startBX = 150
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 48
[ HgcrocEmulator ] 7403 debug:   iADC = 9 at startBX = 175
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   we are adding the hit IFF iSOI= 2's adc_t = 49 >= thresh (53)
```
--> not read out as 49 < 53

```
[ ecalDigis ] 7403 debug: 335675556 0.449145 MeV at 78.0981
[ HgcrocEmulator ] 7403 debug:   iADC = 0 at startBX = -50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 1 at startBX = -25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 2 at startBX = 0
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 3 at startBX = 25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 48
[ HgcrocEmulator ] 7403 debug:   iADC = 4 at startBX = 50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 5 at startBX = 75
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 152
[ HgcrocEmulator ] 7403 debug:   iADC = 6 at startBX = 100
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 73
[ HgcrocEmulator ] 7403 debug:   iADC = 7 at startBX = 125
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 8 at startBX = 150
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 9 at startBX = 175
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   we are adding the hit IFF iSOI= 2's adc_t = 49 >= thresh (53)
``
--> same

```
 ecalDigis ] 7403 debug: 335675480 1.21645 MeV at 78.242
[ HgcrocEmulator ] 7403 debug:   iADC = 0 at startBX = -50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 1 at startBX = -25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 2 at startBX = 0
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 51
[ HgcrocEmulator ] 7403 debug:   iADC = 3 at startBX = 25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 51
[ HgcrocEmulator ] 7403 debug:   iADC = 4 at startBX = 50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 5 at startBX = 75
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 322
[ HgcrocEmulator ] 7403 debug:   iADC = 6 at startBX = 100
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 114
[ HgcrocEmulator ] 7403 debug:   iADC = 7 at startBX = 125
[... 59 lines cut ...]
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 1 at startBX = -25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 2 at startBX = 0
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 3 at startBX = 25
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 49
[ HgcrocEmulator ] 7403 debug:   iADC = 4 at startBX = 50
[ HgcrocEmulator ] 7403 debug:     we are in ADC read-out mode, adc = 50
[ HgcrocEmulator ] 7403 debug:   iADC = 5 at startBX = 75
[ HgcrocEmulator ] 7403 debug:     we are in TOT read-out mode, TOT = 8.63205
[ HgcrocEmulator ] 7403 debug:     Adding TOT hit with toa = 513, tdc_counts = 226 adc_t at prev iADC = 50
[ HgcrocEmulator ] 7403 debug:   Adding further hits with ADC [t-1] = 0x3FF, toa = 0x3FF, until digiToAdd.size() = 6 < nADCs_(10)
[ ecalDigis ] 7403 debug:  The HGCROC will read-out this hit!
```

--> TOT mode, and since we always return `true` for TOT mode, this will be read out, even if the toa = 513

This confirms what Tom said above.
I propose we add in the TOT mode that we only read out if the toa is less than the nADC * clock cycle (i.e. 250)
```
      int toa_max = nADCs_ * clockCycle_;
      return (toa < toa_max);
```
I made a draft PR with that: https://github.com/LDMX-Software/ldmx-sw/pull/1600

tvami, 2025-02-13:
Jeremy wrote in an email thread that the TOT readout should be cut off at "One Bx after nominal."
So I'll have to change the `toa_max`

closed 2025-02-17 by tvami