LDMX-Software/ldmx-sw #1587: Filter for tagging fiducial events

EBerzin, 2025-02-10:
For consistent definitions of signal efficiency across different analyses, we would like to define a clear fiducial volume, and classify events based on this definition.

**Describe the solution you'd like**
We've discussed the following truth-level selections on the recoil electron:

1. |p| < 50 MeV (at production)
2. \> 4 recoil tracker layers with RecoilSimHits
3. \> 0  ECalSimHit

It would be nice to have a filter (or processor) to flag events based on whether or not they satisfy these conditions. 

**Describe alternatives you've considered**
This could also be done at the analysis level, but this leaves room for inconsistency. 

**Additional context**
This is based on discussions at a recent [SWAN meeting](https://indico.fnal.gov/event/68163/contributions/308977/attachments/185117/254756/SWAN02102025.pdf). @tvami @therwig @bryngemark

EBerzin, 2025-02-11:
What is the best approach for this? I think at the SWAN meeting we discussed that we want to keep non-fiducial events. Is there a way to implement a filter that just tags events rather than aborting?

tvami, 2025-02-12:
The way I'd approach this is to have a producer, which sets a boolean (`fiducial_events`) and then

```
  if (fiducial_events) {
    setStorageHint(framework::hint_shouldKeep);
  } else {
    setStorageHint(framework::hint_shouldDrop);
  }
```

Also I think it's useful to have the reverse, like I did in https://github.com/LDMX-Software/ldmx-sw/commit/9bf6acc1544ed7e7ac2834eb5d0a16cccefa6499

This way we can skim the samples depending if this now producer passed or not -- which in the end will act like a filter. But then if we dont want to skim it will not skim, just flag.

tvami, 2025-02-12:
Just to add a clarification, this approach with `hint_shouldKeep` and `hint_shouldDrop` is how we do it in the ecal / hcal veto, as you could see in the commit I linked a second ago too ttps://github.com/LDMX-Software/ldmx-sw/commit/9bf6acc1544ed7e7ac2834eb5d0a16cccefa6499

bloodyyugo, 2025-02-12:
Yes, this is a great way to do  it.

therwig, 2025-02-12:
Another thing to consider is if a single boolean is sufficient or whether we'd like a few booleans, or (better yet?) an integer or bitmask.  The rationale is that, as currently conceived, the fiducially requirement asks for a 'findable track' and an ecal simhit from the electron. In the not-so-distant-future we might also want to use a flag that selects 'findable tracks' and an hcal simhit from the electron, or a findable track and nothing more.  If we already know how we'd likely define these, we could ask the producer to write boolean flags for each, or pack them into a single value.

EBerzin, 2025-02-12:
Thanks for the suggestions!

I think it would be straightforward to include an integer/bitmask.

Another question: where do we want these flags to live? I assume for the ECal/HCal veto these flags are in those respective collections, but it seems like we want something more general for this.

tvami, 2025-02-12:
I think this is the best place:
https://github.com/LDMX-Software/ldmx-sw/tree/trunk/Recon/src/Recon/Skims

tvami, 2025-02-12:
> (even if everything discussed here is for Sim not reco)

Alternatively we could do a skim folder under the `SimCore/src/SimCore`

tvami, 2025-02-16:
@EBerzin I was using the word producer, but I think technically it should be inheriting from `framework::Analyzer`, since it won't produce a new collection

tvami, 2025-02-16:
As for having a filter at generation level, which may be what @bryngemark brought up, I actually made that half a year ago in https://github.com/LDMX-Software/ldmx-sw/pull/1285 

The cxx code being here
https://github.com/LDMX-Software/ldmx-sw/blob/a7a7eb69ab656522b288b288fa31f039ca08eb77/Biasing/src/Biasing/NonFiducialFilter.cxx#L22

In that PR I put the sim actions together to be an ECAL PN, but the filter `class NonFiducialFilter(simcfg.UserAction):` could be applied for anything else

EBerzin, 2025-02-18:
Hi @tvami, I think I may have misunderstood exactly what we want to do with this. In the example you gave with the ECal veto, there is information about the flag saved in an output collection (in the `passesVeto_` variable), in addition to the `hint_shouldKeep` or `hint_shouldDrop` tags. For this, you mentioned we don't want any additional output collections. Does this mean we only do the `hint_shouldKeep`/`hint_shouldDrop` labeling? I think it might be nice to also have some flag accessible in an output collection.

tvami, 2025-02-19:
Yes you are right, let's keep it a producer!

bryngemark, 2025-02-19:
great discussion so far. i'll add my thoughts about the filter.

what i hear we're getting to so far is a bitmask (1 bit or more...) that hides the very basic checks that "fiduciality" would be based on from the user. this has the advantage that the user doesn't need to know them but not much more. 

what i was thinking of was
1. a central definition of fiducial for our baseline analysis. this will likely be different between and real data (based on simhit info or real, somehow chosen-to-be-good data hits). 
    a) a simple isFiducial fulfills this. 
    b) setting a storage hint also fulfills this but only if you skim on it. 
2. if we want to make sure we don't waste time simulating non-fiducial events, filtering is the way to go.
    a) a storage hint could save us from running reconstruction but not simulation
    b) this is probably only a time saver if there are missing hits in the tracker. that situation could still lead to a lot of calorimeter activity. 0 hits in the ecal means we are anyway not spending time on the ecal... so it saves us a bit of hcal work. 

to me it boils down to use case. if a boolean is what we need for now, then let's start there. if info on the why the classification was made is needed (see the values kept in an array for SimpleTrigger as an example) then i'd use that approach to distinguish it from a normal producer adding new objects to the event bus.

therwig, 2025-02-19:
So my impression is that we (likely ~never?) want to skim/filter on this kind of information.  It's only relevant for the signal samples, and we would only be using it to classify the events based on truth-level info.  This concept of truth-level 'fiduciality' as we've defined it doesn't really make sense for background processes or data.

In principle, one could remove the non-fiducial signal events, but this is a relatively small fraction of the total for most masses (so there's no real 'computational savings') and it would be useful to retain the ability to study them more in the future.

There may be some interesting related questions, concerning what reco-level criteria could be used to pick out 'analyzable' events, but I wouldn't tie that discussion to this issue.

PR #1631 (2025-02-27): Processor to flag fiducial recoil electrons
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?

This resolves #1587.

I added a processor to check if the recoil electron in an event satisfies certain fiduciality conditions. The output is saved in a new `FiducialFlag` object, which is structured similarly to `TriggerResult`. It includes booleans for whether the recoil electron has the minimum energy at production (by default 50 MeV), the minimum number of hits in the tracker (by default 5 hits), and has a hit in the ECal/HCal. It also includes an intege [... cut]
closed 2025-02-27 by tvami