LDMX-Software/ldmx-sw #1612: ldmx-sw version is one release behind true value

tomeichlersmith, 2025-02-14:
**Describe the bug**
We are not updating `VERSION` until we make updated gold histograms, but that means the `VERSION` used when building the production image is still behind by one release.

**To Reproduce**
Steps to reproduce the behavior:
1. `denv init ldmx/pro:v4.2.11`
2. `denv fire ../ldmx-sw/SimCore/test/basic.py`
3. `denv config print`

**Desired behavior**
The `ldmx-sw version` and the `denv_image` should align.
I'm not sure how best to do this, but we should do the following steps in order whenever we want to make a release.
1. Choose a new version tag
2. Commit the new version tag to VERSION on trunk
3. Tag this commit as the version
4. Make GitHub Release with this tag
5. Build Production Image and Generate New Gold

Currently, the Generate New Gold workflow and Build Production Image workflow are both run at the same time off the same commit (which is properly tagged according to `git` but not according to `VERSION`) meaning that the production image reads a `VERSION` that is wrong.

**Additional context**
Reported by @jpasc27 on slack

[image]

tvami, 2025-02-14:
Should we not just have VERSION = VERSION + 1 ?

tomeichlersmith, 2025-02-17:
What's `+ 1`? Do we always bump the patch number? What if we want to bump the minor or major number?

These questions are what motivate me to make sure the VERSION file is truly the version. It is only out of date for one commit and I think we can resolve it by doing some more action nonsense.

tvami, 2025-02-21:
@tomeichlersmith will you make the changes for this? I'm not sure how to achieve what you outlined above

tomeichlersmith, 2025-02-21:
Yea, I can put it on my to do list, not sure how quickly I can do it but at least we have this issue documenting the confusion.

tomeichlersmith, 2025-02-24:
So, I tried out different ways of having the actions push a new commit and then update the tag in [a different repo](https://github.com/tomeichlersmith/test-release-repo) and they generally conflict with any protection rules we might impose (like no force-pushing to `trunk`). This makes me think that we could either

1. Use a workflow to "draft a new release" which includes opening the new golden histograms PR. Then the person who triggered that workflow would merge the golden-update PR and then make a new release. I don't like this because it is very error prone - we could easily forget to draft a new release using the workflow and then the golden histograms (and VERSION file) are not updated.
2. Make VERSION more dynamic. Instead of committing it into the source tree, we generate it every time we build. I like this because we can use `just` to make sure this is done correctly using the `git` tags as the single source of truth about what the version is.

Since `git` isn't in the development image (and [we don't want it to be](https://tomeichlersmith.github.io/denv/os-sidebar.html#mac-os-x)), I think adding a `just` recipe that generates the VERSION for us would be the most ergonomic option. Not sure how `git fetch --tags` and `git describe --tags` will behave in the no-depth clones in the CI but I can work on testing that next.

```diff
diff --git a/justfile b/justfile
index 106c92cc..284c5854 100644
--- a/justfile
+++ b/justfile
@@ -82,8 +82,14 @@ configure-clang-lto: (configure "-DENABLE_LTO=ON -DCMAKE_CXX_COMPILER=clang++ -D
 configure-clang-lto-fail-on-warning: (configure "-DENABLE_LTO=ON -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang -DWARNINGS_AS_ERRORS=ON")
 # Keep debug symbols so running with gdb provides more helpful detail
 configure-gdb: (configure-base "-DCMAKE_BUILD_TYPE=Debug")
+
+# prep version file
+[private]
+prep-version:
+    git fetch --tags && git describe --tags | cut -f 1 -d '-' > VERSION
+
 # compile and install ldmx-sw
-build ncpu=num_cpus():
+build ncpu=num_cpus(): prep-version
     denv cmake --build build --target install -- -j{{ ncpu }}
 
 # run the ldmx-sw tests
```

PR #1634 (2025-02-26): 1612 patch version deduction
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1612 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments. This works great in situations where the entire `git` tree is available, but the CI does a sparse/light checkout so `git describe --tags` fails.
- [x] I ran my developments and the following shows that they are successful

from CI https://github.com/LDMX-Software/ldmx-sw/actions/runs/13504011124
```
fatal: No tags can describe 'c4a44984342c2243848539fadbde5f018edf2be9'.
Try --a [... cut]
closed 2025-02-26 by tomeichlersmith