LDMX-Software/ldmx-sw #1660: EcalWABRecProcessor does not find ReducedTracks

tvami, 2025-03-25:
**Describe the bug**
https://github.com/LDMX-Software/ldmx-sw/actions/runs/14065137112/job/39387124993#step:6:420
```
[ LinearRecoilTracksDQM ] 1 debug: Track Monitoring on duplicates and fakes
  [ fire ] 1 fatal: [ProductNotFound] : No product found for name 'ReducedTracks'
    at /home/runner/work/ldmx-sw/ldmx-sw/Framework/include/Framework/Event.h:293 in getObject
  [ fire ] 1 debug:     0 std::vector<ldmx::StraightTrack, std::allocator<ldmx::StraightTrack> > const& framework::Event::getObject<std::vector<ldmx::StraightTrack, std::allocator<ldmx::StraightTrack> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const + 7960 
      1 ecal::EcalWABRecProcessor::produce(framework::Event&) + 1618 
```

**Desired behavior**

I think this is just a renaming issue.

tvami, 2025-03-25:
@SanjitMasanam can you have a look?

I'm not sure how we missed this in the PR @tomeichlersmith

tvami, 2025-03-25:
@tomeichlersmith something is going wrong with the checkout I think, bc in the PR
https://github.com/LDMX-Software/ldmx-sw/actions/runs/13850377142/job/38757152245
this did pass

tomeichlersmith, 2025-03-25:
I don't think the GitHub checkout is doing anything wrong. 

I think its more likely that the GitHub file diff view tries to be "smart" and avoid showing "unrelated changes". GitHub checks out **the branch** and does not apply the changes listed in the "files changed" tab to trunk.

For example, I restored the `iss1499-introduce-wabProcessor` branch and checked it out locally.
```
git fetch
git switch iss1499-introduce-wabProcessor
```
The `git diff` is more interesting than you'd expect. With this branch merged as the latest commit on trunk, I'd naively expect there to be no differences at all.
```
tom@appa:~/code/ldmx/ldmx-sw$ git diff trunk --name-status
M       .github/workflows/asan_ubsan_build.yml
M       .github/workflows/basic_test.yml
M       .github/workflows/compile_clang_lto.yml
M       .github/workflows/docs.yml
M       .github/workflows/format-check.yml
M       .github/workflows/generate_pr_gold.yml
M       .github/workflows/new_pre_release.yml
M       .github/workflows/pr_validation.yml
M       .gitmodules
M       CMakeLists.txt
M       DQM/python/dqm.py
M       DQM/src/DQM/TrkDeDxMassEstFeatures.cxx
M       Ecal/include/Ecal/EcalVetoProcessor.h
M       Ecal/python/vetos.py
M       Ecal/src/Ecal/EcalVetoProcessor.cxx
M       Framework/app/fire.cxx
M       Hcal/src/Hcal/HcalDigiProducer.cxx
M       Recon/include/Recon/Event/TrackDeDxMassEstimate.h
M       Recon/include/Recon/PFTrackProducer.h
M       Recon/include/Recon/TrackDeDxMassEstimator.h
M       Recon/src/Recon/Event/TrackDeDxMassEstimate.cxx
M       Recon/src/Recon/PFTrackProducer.cxx
M       Recon/src/Recon/TrackDeDxMassEstimator.cxx
M       Tracking/CMakeLists.txt
A       Tracking/acts
M       Tracking/include/Tracking/Reco/TruthSeedProcessor.h
M       Tracking/include/Tracking/dqm/TrackingRecoDQM.h
M       Tracking/src/Tracking/Reco/TruthSeedProcessor.cxx
M       Tracking/src/Tracking/dqm/TrackingRecoDQM.cxx
M       cmake/BuildMacros.cmake
```
These differences do not show up in the ["Files changed" tab](https://github.com/LDMX-Software/ldmx-sw/pull/1652/files) because they are from commits that _aren't on that branch_.

I think the reason this is showing up from this branch is because it is affected by commits that were made on `trunk` and were not copied onto this branch. This confusion is why I push folks to learn rebase (but rebase has its own annoyances).

tomeichlersmith, 2025-03-25:
Well, now I'm confusing myself more as well. I really hope we just got our branches tangled with all the developments going on concurrently and its not a bigger deal, but I'm not so sure anymore.

tvami, 2025-03-25:
I thought that 
1) if we use run-validation command it's checking out the branch only
2) if we do the "draft PR --> ready for review" then it does a merge of the branch on the top of the branch.

So if 2) was supposed to catch that in the meantime the `ReducedTracks` was renamed to `StraightTrack

tvami, 2025-03-25:
> This confusion is why I push folks to learn rebase (but rebase has its own annoyances).

Can we used `dependabot` to rebase in the PR automatically?

tomeichlersmith, 2025-03-25:
I think you are correct about what is checked out, not sure how to double check though.

Its not dependabot, but we can enable branch updates either via a button click or automatically. The trouble is then teaching people how to sync their local branch with the remote branch which is not trivial in the case where the remote branch has been rebased and the local branch hasn't.

tomeichlersmith, 2025-03-25:
Got it. There isn't an issue with GitHub's method for checkout (`actions/checkout`) but we (specifically, me) added a sneaky bug when updating the PR Validation.

https://github.com/LDMX-Software/ldmx-sw/blob/073aabe215f3fdeede3c5df036ca44643545a468/.github/workflows/pr_validation.yml#L65-L67

This is meant to copy over the golden histograms and golden logs from `trunk`, but it _also_ copies over the `config.py` from `trunk` meaning that any changes to a `validation_samples/*/config.py` will not be applied within the PR Validation workflow for that PR. For the PR used as an example here, that meant the processor was never included in the config and it was never run.

tvami, 2025-03-25:
Ahhh I see that makes sense! Thanks for finding that!

closed 2025-03-27 by tvami