LDMX-Software/ldmx-sw #1666: Add SimHits clustering algorithm for Hcal

tillrue, 2025-03-27:
**Is your feature request related to a problem? Please describe.**
As part of my bachelor degree project, we discovered that the PF HCal Clustering algorithm doesn't quite work because it is using the HcalRecHits (which seem to have some problems). To get around this, I would like to modify the algorithm such that it can additionally work with the SimHits.

**Describe the solution you'd like**
The additional SimHits feature should be executable from the config and thus changes in the PFHcalClusterProducer and DBScanClusterBuilder are needed. This will probably be some if statements such that the correct methods are used. 

**Describe alternatives you've considered**
The more elegant way would be to directly fix the HcalRecHits instead, but we decided that this would go beyond the scope of my project and having a clustering algorithm that works with the SimHits might still be useful for future development.

**Additional context**

SimHits with extended electron track and HcalRecHits Cluster Centroids
[image]

HcalRecHits for same event
[image]

tvami, 2025-03-28:
What's the Z of these hits? If they are in the side HCAL, this seems pretty realistic. Then I'd suggest to use the orientation information (was added in https://github.com/LDMX-Software/ldmx-sw/pull/1577 ), and if the bar is oriented in X dont use the X coordinate when you do the clustering.

The last line of code is interesting, how can there be a hit if energy = 0? Do you get that with both the simple and original reco?

bryngemark, 2025-03-31:
@tillrue are you using integers for the energy printout that @tvami highlights? in that case the energy might be small but non-zero also fot that hit.

@tvami we'd have to look into the clustering code and maybe the simplified rechit code to see if the bar orientation is taken into account. i think this should be handled at rechit level if it's not already done.

tillrue, 2025-03-31:
Yes, the energy of the hit is actually 0.312494 and not zero.

EinarElen, 2025-04-04:
Apologies in advance for a not particularly well formulated/coherent set of questions, I am very tired when writing this!

If you are getting similar results with the two recproducers, I would suspect that the issue is not in hcal reco but either in the clustering or somewhere else. 

@bryngemark what do you mean with accounting for the bar orientation? 

When it comes to visualizing and understanding Hcal rechits, I think it is always a good idea to have some visualization showing the energy differences between the hits so you can tell the difference between a small hit and the core part. I'm a bit confused as to why there is no distribution along the length of the bar in the figure though?

EinarElen, 2025-04-04:
Right, I'll leave Hcal discussions to elsewhere, but it isn't obvious from the 3d figure that something is wrong. Doing things in sim first is a good idea regardless!

closed 2025-04-09 by tillrue