LDMX-Software/ldmx-sw #1672: Update 3D Hit Construction in reduced-ldmx Tracking

fdelzanno, 2025-04-01:
**Is your feature request related to a problem? Please describe.**

Currently, the reduced tracking scripts only create 3D points with x and z ldmx coordinates, while the y coordinate is always taken to be 0 (since this is the insensitive direction along the strips).

**Describe the solution you'd like**

The tracking should take advantage of the axial-stereo pair geometry to construct the y coordinate and make a representative 3D spatial point in the tracking geometry. This would be similar to the hit construction used in HPS.

**Describe alternatives you've considered**

We initially tried using an energy-weighted average of the global position of (digitized) hits in the tracker, but this is not really a realistic approach to tracking.

fdelzanno, 2025-05-10:
Ok, the reconstruction code is working. I am attaching the output of a 5,000 event, 4 GeV e-gun sample. Here is an overview of the changes:

- Added the 3D hit reconstruction code contributed by @bloodyyugo. This takes an axial and stereo measurement and constructs a 3D hit from the axial projection onto the stereo layer. Currently, it is configured to use sim hits to define the projection onto the stereo layer. 
- Updated the recoil uncertainties based on error propagation of the known uncertainty on the tracker measurements (in the sensitive direction) through the projection we use to define the 3D points. 
- Added a series of histograms to the LinearSeedFinder.cxx that can be used to evaluate the performance of the 3D hit reconstruction. These histograms compare the reconstructed hit's x and y positions to the sim hit's x and y positions at the stereo layer associated with the stereo measurement (the association is done using trackID).

Here are the statistics from the newest sample:

There are 5000 events in the sample.
We find 4948 tracks in the sample, and there are 270 empty events in which we don't find a track
We find 4869 truth tracks in the sample, and there are 131 empty events in which we don't fit a truth track
Out of the tracks in the sample, there are 4769 tracks that have trackID = 1 and 179 tracks that have a different trackID number
Out of the tracks with trackID = 1 in the sample, there are 4702 that have truthProb = 1, 28 tracks that have truthProb = 0.75, 39 tracks that have truthProb = 0.5, and 0 tracks that have truthProb = 0.25.
Out of the tracks in the sample, there are 4857 tracks that corresponds to electrons and 91 tracks that correspond to positrons.

I also added a series of histograms in the LinearSeedFinder.cxx that track how well we are reconstructing the 3D hit. These histograms compare the difference in x- and y-coordinates between reconstructed hit and stereo layer sim hit for both sensors. Here are the resulting histograms, if we only look at axial-stereo hit combinations that have the same trackID.
For Sensor 1 (i.e. the first axial-stereo pair):

[image]

[image]

For Sensor 2 (i.e. the second axial-stereo pair):

[image]

[image]

and here are the resulting histograms, if we look at ALL the axial-stereo hit combinations that we can generate by combining an axial hit with a stereo hit:
For Sensor 1 (i.e. the first axial-stereo pair):

[image]

[image]

For Sensor 2 (i.e. the second axial-stereo pair):

[image]

[image]

fdelzanno, 2025-05-11:
Here are the comparisons between 5,000 event, 4 GeV e-gun samples generated with our old rTracking (red) (where we just set y=0 and essentially only did the fitting in x) and the new rTracking that reconstructs 3D points (blue). The main difference is that we did not have any constraint on positions in y in the old sample and had a high uncertainty in that coordinate.

[image]

This is the track's distance to the nearest ECalRecHit. This change makes sense. In the previous sample, we had such a high uncertainty in y that the fit could tune itself to be much closer to the RecHit. Now, the uncertainty bounds the track, and we see more of a spectrum in distance.

[image]

By similar reasoning, we expect the Chi-2 of the tracks to increase in the new samples. 

[image]

The number of tracks is more or less consistent -- with more empty tracks in the new sample (since our uncertainty provides a greater constraint on potential track candidates).

Here are the pulls for the angular variables:

[image]

[image]

We see a major improvement in theta. This makes sense, since theta is derived from the y coordinate (which we only knew with great uncertainty in the old fitting). 

Here are the differences in track position at the target:

[image]

[image]

Again, we see a major improvement for loc1 (which corresponds to y).

Here are the differences in track position at the ECal:

[image]

[image]

There is less of an improvement in loc1 at the ECal. This makes sense. The third reference point for the fit is the RecHit, so the large uncertainty in the old case allows the track to match with the ECal much better than with the target.

Finally, here is the PID information:

[image]

tvami, 2025-05-13:
@fdelzanno I think this looks pretty good! Will you open a PR next?

PR #1705 (2025-05-14): Add Hit Reconstruction from Axial and Stereo Tracking Layers to rLDMX Tracking
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1672 by adding a method to LinearSeedFinder.cxx that reconstructs 3D hits from all combinations of an axial and a stereo measurement. It also updates the uncertainties associated with the x and y coordinates of the reconstructed hit. 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments.
- [x] I read, understood and follow the [coding rules](https://ldmx-software.github.io/developing/cod [... cut]