LDMX-Software/ldmx-sw #1681: Fix SimpleDigiAndRecProducer.cxx and implement Side Hcal Algorithm

tillrue, 2025-04-07:
**Describe the bug**
When using the SimpleDigiAndRecProducer.cxx to produce RecHits, I noticed that these do not resemble the SimHits very well. Further, the algorithm for the Side Hcal was not implemented yet and used the center of each bar (in all directions) where the SimHits were located.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the SimpleDigiAndRecProducer to obtain RecHits
2. Inspect RecHits and SimHits and notice a discrepancy

**Desired behavior**
The SimpleRecHits should give a reconstructed picture of the SimHits for each event.

**Screenshots**
This screenshot shows the SimHits of a wide scattering event with a shower in the Side Hcal. Note that the left front side represents the ecal/side hcal front on which the beam is incident.

[image]

This screenshot shows the current SimpleRecHits of the same event in the Side Hcal (when removing the restriction to the center of each bar in the side hcal). If we compare this to the simhits, we see that they are much further spread out. The size & colour of the boxes represent the deposited energy.

[image]

This screenshot shows the new/fixed SimpleRecHits of the same event in the Side Hcal without smearing. Comparing this to the Simhits (first picture), we can see that they are much better resembled.

[image]

EinarElen, 2025-04-07:
Yep this is a straight-up bug from when #1591 was merged where the priority I think was getting energy resolution... 

If any of the regular ldmx-sw developers is able to, it would be good to include some checks (in addition to the  figures above of course) either when merging this (if @tillrue/others has no issue working on their local version) or as a separate issue (if @tillrue/others needs it). I would recommend the latter :)

tillrue, 2025-04-07:
Sure, I have no issue with working on my local version to continue my project. Just wanted to mention this issue and the "relatively simple" fix as it might be the source of problem for other people as well. My changes could probably be improved as I'm just starting to develop and should definitely be checked by someone with more experience!

EinarElen, 2025-04-07:
Your fix is almost certainly just right, from what I can tell it should be correct, but especially with geometry-code it is my experience to never ever trust what looks correct. I would still recommend we merge this and make a separate issue for setting up some checks (I'm unfortunately not  available at the moment for working on that issue).

closed 2025-04-09 by tillrue