LDMX-Software/ldmx-sw #1686: Move to the "new" simple HCAL reco in CI

tvami, 2025-04-10:
**Is your feature request related to a problem? Please describe.**

There are several problems with the "old" HCAL reco

**Describe the solution you'd like**

Chatting with @bechenard I understand we should really just move to the simplified reco introduced (back) in https://github.com/LDMX-Software/ldmx-sw/pull/1591

tillrue, 2025-04-10:
Hi! I think this is a good idea!

When I was looking at the output of the simple hcal reconstruction, i noticed that the smearing was quite big due to the position_resolution set to 150 mm by default. Does anyone know if these 150 mm are chosen to specifically meet the construction parameters of the HCal or could I in theory also try to reduce it a bit to improve the results?

EinarElen, 2025-04-10:
The 150mm is taken from the old version (https://github.com/LDMX-Software/Hcal/blob/ddaade2b19741521938194c7d8d4968a617a3511/include/Hcal/HcalOldDigiProducer.h) which I _think_ was the version used for the whitepaper, but maybe @cmantill or @tomeichlersmith knows?

tomeichlersmith, 2025-04-10:
While I am not opposed to re-adopting the simplified digi/reco chain for the HCal, I would like there to be an issue (or issues) documenting the problems with the more complicated chain. Do those issues already exist and I am just being lazy?

tvami, 2025-04-10:
The way I ran into this is that I wanted this plot for the DR:
[image]

So we have these big peak at 250 and a cut-off at 500. 
Now looking at the plots in https://github.com/LDMX-Software/Hcal/pull/75 I see the same.
So I guess that PR has the issues "listed" in plots?

tomeichlersmith, 2025-04-10:
Thanks 👍 I just want to make sure we can find a "To Do" list to start from in the future.

bechenard, 2025-04-10:
Regarding the 150mm, as Tamas pointed, this is the resolution along the bar. While we should be able to (theoretically) do better, the test beam analysis points to a value close to this at the moment, so this is a reasonable starting point. This said, we haven't really used this information so far, so I would keep it as is until we have some solid measurements to support a better value.

closed 2025-04-10 by tvami