LDMX-Software/ldmx-sw #1694: storage location for golden histograms and logs

tomeichlersmith, 2025-04-14:
We have a functional solution right now, but it has a few issues.

1. We have a hard upper limit on the size of our golden files (100MB I believe?).
2. The histogram files are binary files and so each new version of them is committed into the `git` tree into a fully new object which increases the size of the `git` repo slowing the clone down considerably.

In a perfect world, neither of these issues would be present. The golden files would reside on the machine(s) that run the CI and we would only interact with them directly if something with the CI was going wrong. A normal developer would not need to copy them down to their machine and the only size limit on these golden files would be dictated by the disk space of the machine(s) that run the CI.

This issue is just me documenting some ideas while I'm thinking about it. I doubt that this is high enough priority to resolve anytime soon.

## Self-Hosted Runners
The ideal solution is for us to administer the machines that run the CI with the obvious downside that we would need to bear the additional burden of administration. Nevertheless, I still think this is a reasonable solution since we already have experience with administering clusters for large-scale simulation production. In fact, the machine that builds the ldmx/dev images is already a [self-hosted runner at UMN](https://ldmx-software.github.io/dev-build-context/runner.html).

### System Requirements
- Pre-installed Dependencies (`apptainer`, `just`, `denv`, other helpers...)
- Sharing golden files across machines either via a shared filesystem or manual duplication
- Access to update the golden files from the runners themselves (`update-pr-gold` job)
- 16+ cores (2 cores per job, allows the PR Validation jobs to be run in parallel)
- 1TiB+ (~64GiB per job is plenty if jobs are _sharing_ image cache and _sharing_ the golden files)
- 32GiB+ (2GiB per job)

### Additional Features if Possible
- Local image cache to save time on jobs
- `docker` install for building and pushing the production images
- Storage of full golden event files for precise comparison (not just histogram KS test but starting with a full event-by-event diff)
- Local repo cache to save time on jobs (have GitHub's actions/checkout do the checkout logic for us but not remove afterwards? not sure if possible...)

## Separate Repo for Golden Files
The only other idea I can think of is putting the golden files into their own `git` repository which we then only push/pull from in CI when necessary. This would then only waste time moving the large `git` objects in the CI and not for the normal human developer.

tvami, 2025-04-14:
I think having an `ldmx-data` repository would be great, I could also imagine the LHE files for the signal to be kept there, so we dont need to rerun the MG all the time, even tho that does not change very often (like changes every 5 years when we update the version? lol)

tvami, 2025-06-06:
> That's a decrease of almost 4x !!!

really nice!!

tomeichlersmith, 2025-06-10:
Resolved by #1721 , discussion of pruning the history is moved to #1726

closed 2025-06-10 by tomeichlersmith