LDMX-Software/ldmx-sw #1707: Potential out-of-bounds memory in IdealClusterBuilder

tvami, 2025-05-16:
**Describe the bug**

In https://github.com/LDMX-Software/ldmx-sw/actions/runs/15058735763/job/42329736694
ASAN gives us 

```
[ 94%] Building CXX object TrigScint/CMakeFiles/TrigScint.dir/src/TrigScint/QIEInputPulse.cxx.o
In file included from /usr/include/c++/13/algorithm:60,
                 from /home/runner/work/ldmx-sw/ldmx-sw/Trigger/Algo/include/Trigger/IdealClusterBuilder.h:4,
                 from /home/runner/work/ldmx-sw/ldmx-sw/Trigger/Algo/src/Trigger/IdealClusterBuilder.cxx:1:
In static member function 'static constexpr _Up* std::__copy_move<_IsMove, true, std::random_access_iterator_tag>::__copy_m(_Tp*, _Tp*, _Up*) [with _Tp = const trigger::Hit; _Up = trigger::Hit; bool _IsMove = false]',
    inlined from 'constexpr _OI std::__copy_move_a2(_II, _II, _OI) [with bool _IsMove = false; _II = const trigger::Hit*; _OI = trigger::Hit*]' at /usr/include/c++/13/bits/stl_algobase.h:506:30,
    inlined from 'constexpr _OI std::__copy_move_a1(_II, _II, _OI) [with bool _IsMove = false; _II = const trigger::Hit*; _OI = trigger::Hit*]' at /usr/include/c++/13/bits/stl_algobase.h:533:42,
    inlined from 'constexpr _OI std::__copy_move_a(_II, _II, _OI) [with bool _IsMove = false; _II = const trigger::Hit*; _OI = trigger::Hit*]' at /usr/include/c++/13/bits/stl_algobase.h:540:31,
    inlined from 'constexpr _OI std::copy(_II, _II, _OI) [with _II = const trigger::Hit*; _OI = trigger::Hit*]' at /usr/include/c++/13/bits/stl_algobase.h:633:7,
    inlined from 'constexpr void std::vector<_Tp, _Alloc>::_M_assign_aux(_ForwardIterator, _ForwardIterator, std::forward_iterator_tag) [with _ForwardIterator = const trigger::Hit*; _Tp = trigger::Hit; _Alloc = std::allocator<trigger::Hit>]' at /usr/include/c++/13/bits/vector.tcc:341:15,
    inlined from 'constexpr void std::vector<_Tp, _Alloc>::_M_assign_aux(_ForwardIterator, _ForwardIterator, std::forward_iterator_tag) [with _ForwardIterator = const trigger::Hit*; _Tp = trigger::Hit; _Alloc = std::allocator<trigger::Hit>]' at /usr/include/c++/13/bits/vector.tcc:315:7,
    inlined from 'constexpr std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = trigger::Hit; _Alloc = std::allocator<trigger::Hit>]' at /usr/include/c++/13/bits/stl_vector.h:790:21,
    inlined from 'void trigger::IdealClusterBuilder::Build2dClusters()' at /home/runner/work/ldmx-sw/ldmx-sw/Trigger/Algo/src/Trigger/IdealClusterBuilder.cxx:215:27:
/usr/include/c++/13/bits/stl_algobase.h:437:30: warning: 'void* __builtin_memmove(void*, const void*, long unsigned int)' reading between 45 and 9223372036854775807 bytes from a region of size 44 [-Wstringop-overread]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/runner/work/ldmx-sw/ldmx-sw/Trigger/Algo/src/Trigger/IdealClusterBuilder.cxx: In member function 'void trigger::IdealClusterBuilder::Build2dClusters()':
/home/runner/work/ldmx-sw/ldmx-sw/Trigger/Algo/src/Trigger/IdealClusterBuilder.cxx:215:27: note: source object '<anonymous>' of size 44
  215 |       layer_hits[l] = {hit};
      |  
```

**Desired behavior**

Clean compilation

tvami, 2025-05-16:
Could be false-postive? @EinarElen wdyt?

EinarElen, 2025-05-16:
False positives from asian are really really uncommon so I'd be surprised

EinarElen, 2025-05-16:
Oh wait, this is a compilatio

tvami, 2025-05-16:
Yes it is compiler, but I noticed it in the ASAN build

therwig, 2025-08-04:
Yes, I wrote it a while ago and can take a look if its generating some errors.