LDMX-Software/ldmx-sw #1709: using `clang-tidy` for coding rules and linting

tomeichlersmith, 2025-05-19:
`clang-tidy` can go a bit further than `clang-format`, enforcing many of our [Coding Rules](https://ldmx-software.github.io/developing/coding-rules.html) (e.g. naming conventions) as well as provide some helpful preventative measures to avoid common issues that can be found via static analysis.

## Notes
- CMake generates a "compile command database" for us, which is required by `clang-tidy` to function. Use the `-p` argument to provide the build directory.
- Like `clang-format`, we need to provide the files to tidy. Probably could do something similar to `just format-cpp` where we write a file listing using `find`
- Looks in current directory's `.clang-tidy` file for configuration.
- Provide `--fix` argument to apply the fix (if possible)
- `--warnings-as-errors` argument to fail on CI?

## Config
Scaffolding up the YAML to go into a `.clang-tidy` file
```yaml
# comma-separated list of globs to enable or disable (`-` prefix)
# this one disables all and then enables readability-identifier-naming
Checks: "-*,readability-identifier-naming"
# options to checks
CheckOptions:
  readability-identifier-naming.FunctionCase: camelBack
  readability-identifier-naming.NamespaceCase: lower_case
  readability-identifier-naming.ClassCase: CamelCase
  readability-identifier-naming.StructCase: CamelCase
  readability-identifier-naming.VariableCase: lower_case
  readability-identifier-naming.GlobalConstantCase: UPPER_CASE
  readability-identifer-naming.MemberCase: lower_case
  readability-identifier-naming.MemberSuffix: "_"
  readability-identifer-naming.ConstantMemberCase: UPPER_CASE
```

tvami, 2025-08-06:
I tried to add more "targets" in the `CMakeFile` but that didnt fix it. I also tried to add `-I <all the includes>` directly to `clang-tidy`, that also didnt help

tvami, 2025-08-06:
Yeah but what's even more weird is when I add the `.clang-tidy` file, the normal compiler will give warnings (and suggestions), so it works in that direction, but then the compiler doesnt have the fix option and then at `clang-tidy` it doest work anymore. Which also makes it so that I cant add `.clang-tidy` otherwise our warnings-as-error build will fail for the cosmetics....

tomeichlersmith, 2025-08-06:
Yea, we probably need to look through our CMake options to see if we can disable default-running `clang-tidy` during compilation. If `CMAKE_CXX_CLANG_TIDY` is set, then `clang-tidy` is run during the build and (I'm guessing) without a default `.clang-tidy` configuration file it just does nothing.

tvami, 2025-08-06:
I guess this could work after everything is fixed... and after that we wouldnt even need to have another CI since the errors-as-warnings would find all those issues. What I'm really missing is the fact that the clang-tidy doesnt work, bc if did then we could have the CI make the fixes automatically

tomeichlersmith, 2025-08-06:
Not all errors/warnings that are detectable by `clang-tidy` are "fixable" because many of them require a human designer with a more advanced knowledge of the code to understand how to refactor it. For this reason, I don't think having a tidy be auto-applied like the formatting will be possible. Instead, having it be more like a test where humans can inspect it for what to change is more appropriate.

tvami, 2025-08-08:
> We could remove that from the default build config and instead have a spearate recipe focused on running clang-tidy manually

I did this in https://github.com/LDMX-Software/ldmx-sw/pull/1803/commits/84eb46a623dfae0a684d8fdf3f137987890ca79c

tvami, 2025-08-12:
I think I figured it out.
Or at least this
https://github.com/LDMX-Software/ldmx-sw/blob/4c6353bb65a56b41d9b6c323be4002a55a52c5d0/justfile

worked for https://github.com/LDMX-Software/ldmx-sw/pull/1813/

closed 2025-08-20 by tvami