LDMX-Software/ldmx-sw #1717: Set timestamp for ACTS URL

tvami, 2025-05-24:
**Is your feature request related to a problem? Please describe.**

when compiling I see this
```
[ WARNING ]: The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy ;CMP0135 is not set. The policy's OLD behavior will be used. ;When using a URL download, the timestamps of extracted files ;should preferably be that of the time of extraction, otherwise ;code that depends on the extracted contents might not be ;rebuilt if the URL changes. The OLD behavior preserves the ;timestamps from the archive instead, but this is usually not ;what you want. Update your project to the NEW behavior or ;specify the DOWNLOAD_EXTRACT_TIMESTAMP option with a value of ;true to avoid this robustness issue.
```

**Describe the solution you'd like**
I think what we need to do is
```
FetchContent_Declare(
    Acts
    URL https://github.com/acts-project/acts/archive/refs/tags/v36.0.0.tar.gz
    URL_HASH MD5=f543dd8ba030bea2e4ee2f1b07dbe7c0
)
```
to
```
FetchContent_Declare(
    Acts
    URL https://github.com/acts-project/acts/archive/refs/tags/v36.0.0.tar.gz
    URL_HASH MD5=f543dd8ba030bea2e4ee2f1b07dbe7c0
    DOWNLOAD_EXTRACT_TIMESTAMP TRUE
)
```
in
https://github.com/LDMX-Software/ldmx-sw/blob/615bde3eb3e57c7b7f2343b02adbfcd000448a11/Tracking/CMakeLists.txt#L69-L73

tvami, 2025-05-24:
@tomeichlersmith do you agree that the "TRUE" is the correct way to go?

tomeichlersmith, 2025-05-26:
Yea I think TRUE is the way to go.

As a side note, this should only happen with ldmx/dev images before v5 and those images will not be able to build ldmx-sw v4.3.1 or later due to a few missing headers. (@sophiemiddleton and @bryngemark originally found this out and reported it on slack, I've since documented it on the compatibility page: https://ldmx-software.github.io/developing/compatibility.html#-431)