LDMX-Software/ldmx-sw #1729: Add error bars in the plots for Validation module

tvami, 2025-06-14:
**Is your feature request related to a problem? Please describe.**

We went thru this already in the CI comparison plots, we should show error bars on the histograms that we are comparing. 

**Describe the solution you'd like**

Add histo error bars in the Validation module

tomeichlersmith, 2025-06-17:
We should just use [`hist`](https://hist.readthedocs.io/en/latest/index.html) instead of doing it ourselves with `plt.hist`.

The `hist` package interfaces with [`mplhep`](https://mplhep.readthedocs.io/en/latest/) and [`boost_histogram`](https://boost-histogram.readthedocs.io/en/latest/index.html) and does the things we want to do (good uncertainty propagation including weights, sane defaults for error bars, etc...)

I did not know of `hist` when I originally scaffolded up this package and we should not re-invent the wheel.

In fact, I would argue we should just replace all of the dependencies with [`scikit-hep`](https://github.com/scikit-hep/scikit-hep/) which includes all of the dependencies we currently list as well as `hist`, `mplhep`, etc. in addition to others that we will probably find helpful in the future (`awkward`).

https://github.com/LDMX-Software/ldmx-sw/blob/12714e7e1c52fa9abd2cd09ad0a3b93da47ded48/Validation/pyproject.toml#L12-L15

tomeichlersmith, 2025-07-30:
> So now the zero bins have an uncertainty too, which we should not have

I disagree, we do not know that a bin has zero entries with absolute certainty. We could discuss how best to estimate an empty bin's uncertainty. The default mplhep method is to use the near-est non-zero bin to set the scale of a Poisson uncertainty. This works well for smoothly-falling distributions. If we really want to hide the error bars on empty bins (again, I disagree with this), but we can define method that just sets the error on empty bins to zero. See: https://github.com/scikit-hep/mplhep/issues/429

> binning is super busy

I think this is just a relic of the binning being very fine. It looks like the bin widths are on the order of the line width meaning that drawing a vertical bar for the error bars in each bin ends up filling all the bins with color.

tvami, 2025-07-30:
Yes I guess I can be convinced to have the errors on the zero...

> I think this is just a relic of the binning being very fine. It looks like the bin widths are on the order of the line width meaning that drawing a vertical bar for the error bars in each bin ends up filling all the bins with color.

OK but I thought we can rebin in the Validation module, but maybe I lost that functionality