LDMX-Software/ldmx-sw #1763: Move to 400um Si thickness in Ecal

tvami, 2025-07-09:
**Is your feature request related to a problem? Please describe.**

Move to 400um Si thickness in Ecal

**Describe the solution you'd like**

This will make S/N better

tvami, 2025-08-08:
I changed `SensDetThickness` to 0.4 and introduced v15.

```
denv python3 ecal_layer_stack.py ldmx_ecal_v14
```
gives me
```
Layer      dE     X0 Lambda   Zpos
    1   2.009  0.034  0.014  7.582
    2   3.931  0.195  0.024 16.162
    3   6.141  0.480  0.034 33.426
    4   7.245  0.623  0.039 43.506
    5   8.350  0.766  0.044 60.770
    6  10.008  0.980  0.052 71.850
    7  10.671  1.065  0.055 90.614
    8  10.671  1.065  0.055 102.294
    9  10.671  1.065  0.055 121.058
   10  10.671  1.065  0.055 132.738
   11  10.671  1.065  0.055 151.502
   12  10.671  1.065  0.055 163.182
   13  10.671  1.065  0.055 181.946
   14  10.671  1.065  0.055 193.626
   15  10.671  1.065  0.055 212.390
   16  10.671  1.065  0.055 224.070
   17  10.671  1.065  0.055 242.834
   18  10.671  1.065  0.055 254.514
   19  10.671  1.065  0.055 273.278
   20  10.671  1.065  0.055 284.958
   21  10.671  1.065  0.055 303.722
   22  10.671  1.065  0.055 315.402
   23  10.671  1.065  0.055 334.166
   24  14.538  1.565  0.073 345.846
   25  18.295  2.050  0.090 368.110
   26  18.295  2.050  0.090 383.190
   27  18.295  2.050  0.090 405.454
   28  18.295  2.050  0.090 420.534
   29  18.295  2.050  0.090 442.798
   30  18.295  2.050  0.090 457.878
   31  18.295  2.050  0.090 480.142
   32  18.295  2.050  0.090 495.222
   33  18.295  2.050  0.090 517.486
   34   9.743  1.033  0.049 532.566
  Sum 408.017 42.238  2.073 539.898
 Back   1.634  0.036  0.011 539.898
```

And for v15 I get
```
Layer      dE     X0 Lambda   Zpos
    1   2.009  0.034  0.014  7.582
    2   3.931  0.195  0.024 16.162
    3   6.141  0.480  0.034 33.426
    4   7.245  0.623  0.039 43.506
    5   8.350  0.766  0.044 60.770
    6  10.008  0.980  0.052 71.850
    7  10.671  1.065  0.055 90.614
    8  10.671  1.065  0.055 102.294
    9  10.671  1.065  0.055 121.058
   10  10.671  1.065  0.055 132.738
   11  10.671  1.065  0.055 151.502
   12  10.671  1.065  0.055 163.182
   13  10.671  1.065  0.055 181.946
   14  10.671  1.065  0.055 193.626
   15  10.671  1.065  0.055 212.390
   16  10.671  1.065  0.055 224.070
   17  10.671  1.065  0.055 242.834
   18  10.671  1.065  0.055 254.514
   19  10.671  1.065  0.055 273.278
   20  10.671  1.065  0.055 284.958
   21  10.671  1.065  0.055 303.722
   22  10.671  1.065  0.055 315.402
   23  10.671  1.065  0.055 334.166
   24  14.538  1.565  0.073 345.846
   25  18.295  2.050  0.090 368.110
   26  18.295  2.050  0.090 383.190
   27  18.295  2.050  0.090 405.454
   28  18.295  2.050  0.090 420.534
   29  18.295  2.050  0.090 442.798
   30  18.295  2.050  0.090 457.878
   31  18.295  2.050  0.090 480.142
   32  18.295  2.050  0.090 495.222
   33  18.295  2.050  0.090 517.486
   34   9.743  1.033  0.049 532.566
  Sum 408.017 42.238  2.073 539.898
 Back   1.634  0.036  0.011 539.898
```

tvami, 2025-08-08:
I mean I didnt expect this to have a big effect at all, but this has no effect at all, does that make sense to you @tomeichlersmith ?

tomeichlersmith, 2025-08-08:
I should mention that while the layer weights (`dE`) should not include the sensitive silicon, we maybe want to include it in the `Zpos` calculation? I'm not sure where it is used. Personally, I've only used the `dE` column elsewhere.

The GDML only needs the arrays of tungsten thickness because everything else is summed up within the GDML using the constants.

tvami, 2025-08-08:
Gotcha thanks!