LDMX-Software/ldmx-sw #1792: Remove EcalTrackingGeometry

tvami, 2025-07-31:
**Is your feature request related to a problem? Please describe.**

I noticed here 
https://github.com/LDMX-Software/ldmx-sw/blob/28fdf455055fd6b69e958b8779ae8c8248a564df/Tracking/src/Tracking/geo/EcalTrackingGeometry.cxx#L42
that the `EcalTrackingGeometry` is using the v12 geometry.
However, I dont think this is actually used anywhere.

**Describe the solution you'd like**

If not used, remove this class. If used, fix it to use the correct geometry.

tvami, 2025-08-01:
> > However, I dont think this is actually used anywhere.
> 
> Right, [@bloodyyugo](https://github.com/bloodyyugo) ?

Seems it works https://github.com/LDMX-Software/ldmx-sw/pull/1793

tomeichlersmith, 2025-08-01:
The point of the `EcalTrackingGeometry` was to show that we _could_ run Acts tracking within the Ecal _if_ we can load the Ecal geometry into Acts. Other tracking algorithms have been implemented for the Ecal hits. As you can see, this has never really been tested, but the point is that PF when originally developing the tracking checked that the Ecal geometry could also be loaded for this use case. Not sure if the history makes a difference on if it should be removed.

tomeichlersmith, 2025-08-01:
Yea I think that works 👍

closed 2025-08-01 by tvami