LDMX-Software/ldmx-sw #1804: CI fails with new image

tvami, 2025-08-08:
**Describe the bug**

https://github.com/LDMX-Software/ldmx-sw/actions/runs/16836289903
```
Status: Downloaded newer image for ldmx/dev:latest
  docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: exec: "/home/runner/work/ldmx-sw/ldmx-sw/.github/validation_samples/kaon_enhanced": is a directory: unknown: permission denied
```

**To Reproduce**
The CI in `trunk`: https://github.com/LDMX-Software/ldmx-sw/actions/runs/16836289903

**Desired behavior**

tvami, 2025-08-08:
@tomeichlersmith is this because of https://github.com/LDMX-Software/dev-build-context/pull/138 ?

tomeichlersmith, 2025-08-08:
After a brief investigation, the issue is that I removed the entrypoint definition in the image (mistakenly assuming everyone had moved to using `denv` which defines the entrypoint for us), but the CI still uses the old entrypoint-in-image assumption. Updating the CI to use `denv` will fix this issue and align the CI with how we run locally.

tvami, 2025-08-09:
The generate gold still doesnt work
```
 Ref Dir: /home/runner/work/ldmx-sw/ldmx-sw/ci-data/target_genie
  Sample Name: target_genie
  /home/runner/work/ldmx-sw/ldmx-sw/./.github/actions/validate/validate.sh: line 30: denv: command not found
  Sample Dir: /home/runner/work/ldmx-sw/ldmx-sw/.github/validation_samples/target_genie
  Not Running Comparison? true
```

https://github.com/LDMX-Software/ldmx-sw/actions/runs/16853134021/job/47742546645

PR #1805 (2025-08-09): remove `ldmx` in favor of using `denv` in PR Validation
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1804 

While this isn't a full refactoring of the CI, it is a good first draft to test run replacing the entrypoint-based running with the `denv`-based running.
closed 2025-08-11 by tomeichlersmith