LDMX-Software/ldmx-sw #1823: Issues when compiling on Apple Si

tvami, 2025-08-20:
**Is your feature request related to a problem? Please describe.**
```
500.1 gmake[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libblas.so.3', needed by 'SimCore/libSimCore_Generators.so'.  Stop.
500.1 gmake[1]: *** [CMakeFiles/Makefile2:1835: SimCore/CMakeFiles/SimCore_Generators.dir/all] Error 2
```

and I also see a warning with
```
11.84 [  7%] Building CXX object ReconEvent/CMakeFiles/Recon_Event.dir/src/Recon/Event/HgcrocPulseTruth.cxx.o
12.12 In file included from /usr/include/c++/13/bits/stl_algo.h:61,
12.12                  from /usr/include/c++/13/functional:67,
12.12                  from /usr/local/include/root/TF1.h:25,
12.12                  from /code/Recon/include/Recon/Event/CompositePulse.h:6,
12.12                  from /code/Recon/include/Recon/Event/HgcrocPulseTruth.h:8,
12.12                  from /code/Recon/src/Recon/Event/HgcrocPulseTruth.cxx:1:
12.12 /usr/include/c++/13/bits/stl_heap.h: In function 'constexpr void std::__adjust_heap(_RandomAccessIterator, _Distance, _Distance, _Tp, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<pair<double, double>*, vector<pair<double, double> > >; _Distance = long int; _Tp = pair<double, double>; _Compare = __gnu_cxx::__ops::_Iter_comp_iter<ldmx::HgcrocPulseTruth::getMax() const::<lambda(const pair<double, double>&, const pair<double, double>&)> >]':
12.12 /usr/include/c++/13/bits/stl_heap.h:224:5: note: parameter passing for argument of type 'std::pair<double, double>' when C++17 is enabled changed to match C++14 in GCC 10.1
12.12   224 |     __adjust_heap(_RandomAccessIterator __first, _Distance __holeIndex,
12.12       |     ^~~~~~~~~~~~~
```

**Describe the solution you'd like**

The first one is the real issue, and it seems it's manually added in the GENIE here
https://github.com/LDMX-Software/ldmx-sw/blob/65c6b5c984707aee650121ca9316e7f21d6c0889/cmake/BuildMacros.cmake#L136-L138

**Describe alternatives you've considered**

We either add the BLAST to the Docker, or remove it's use.

The second one seems to be solved by moving to `ranges::sort`

tomeichlersmith, 2025-08-20:
You should be able to `just compile`. `just compile` uses the `ldmx/dev` image which has an ARM build. You only need to build the image yourself if you want a `ldmx/pro` image with ldmx-sw already-built inside. These second images do not have ARM builds (yet).

tomeichlersmith, 2025-08-20:
This is fascinating, thank you for reporting back