LDMX-Software/ldmx-sw #1849: v15 production software dev plan

tvami, 2025-09-22:
This is a high level issue collecting what software changes we need for the v15 production

 * Fixing HCAL reco
 * ECAL layer weight update to the 32 layer situation
 * Shall we consider a new ACTS update?
 * Realistic readout for Tracker?
 * Adding TS QIE reco?

EinarElen, 2026-01-15:
Hi everyone! 

I think i might have solutions for some old HCal issues (i don't have the time at the moment to set up a full PR / test everything though... all of this is speculative but should be straightforward to test?)

#1344 #1348 

In  HcalDoubleEndRecProducer, we group single-ended hits by bar ID and attempts to pair positive and negative end hits. The pairing logic initializes indices to `(-1, -1)`:

```cpp
std::pair<int, int> indices(-1, -1);
```

These indices are only updated when matching hits are found:
```cpp
if (digi_id.isNegativeEnd() && indices.second == -1) {
    indices.second = i_hit;
}
if (!digi_id.isNegativeEnd() && indices.first == -1) {
    indices.first = i_hit;
}
```
So.... guess what could go wrong here? :) 

If both sides don't get found, we'd end up with -1 for one of the indices. Later, we end up doing [] access on a container with the indices -> reading junk memory for one half. 

Next, and possibly caused by the previous bug? If one side has invalid TOA (or TOA 0) then the shift gets applied despite that 
```cpp
if (pos_time != 0 || neg_time != 0) {  // OR logic - too permissive
    neg_time = neg_time - mean_shift;
}
```

Basically, if `pos_time = 0` (invalid) and `neg_time = 85 ns` (valid), `hit_time_diff ~ -85 ns` =>  Position = `187.4 * 85 / 2 ≈ 7965 mm` (hilariously bad)

Cheers!

tvami, 2026-01-17:
Maybe we should discuss this in the dedicated HCAL issue. But I've applied a fix for this and does not fix the original issue of having way too much PE at ~250. Looking at the code more in details now, I'm wondering if voltage should have the attenuation correction applied? Currently it's not applied.

tvami, 2026-01-17:
I've continued in https://github.com/LDMX-Software/ldmx-sw/issues/1344

EinarElen, 2026-01-18:
PE at the saturation limit isn't a bug but a reconstruction deficiency, this is resolving a suite of regular old bugs :)