LDMX-Software/ldmx-sw #1853: Acts ParticleHypothesis assert fail

erik-wallin, 2025-09-24:
**Describe the bug**
Acts::GenericParticleHypothesis expects positive PDG IDs as an argument, but if a particle with negative PDG ID enters the TruthSeedProcessor https://github.com/LDMX-Software/ldmx-sw/blob/25e45c03a85d33f73823c7eb8af89a6e2f39925b/Tracking/src/Tracking/Reco/TruthSeedProcessor.cxx#L147 the assert fails:
```
fire: /usr/local/include/Acts/EventData/GenericParticleHypothesis.hpp:52: Acts::GenericParticleHypothesis<charge_t>::GenericParticleHypothesis(Acts::PdgParticle) [with charge_t = Acts::AnyCharge]: Assertion `absPdg == makeAbsolutePdgParticle(absPdg) && "pdg is expected to be absolute"' failed.
Aborted (core dumped)
```

This also applies here: https://github.com/LDMX-Software/ldmx-sw/blob/25e45c03a85d33f73823c7eb8af89a6e2f39925b/Tracking/src/Tracking/Reco/VertexProcessor.cxx#L116

I guess the solution is as simple as flipping the sign? The charge of the particle is given as a separate argument anyway. Don't know if there are any actual consequences to this bug, it just stops me from running with the debugger.

tvami, 2025-09-24:
@bloodyyugo FYI

bloodyyugo, 2025-09-25:
Yeah, ParticleHypothesis takes the absolute value of the pdgid.  

In you can just delete this line in TruthSeedProcessor:

[https://github.com/LDMX-Software/ldmx-sw/blob/25e45c03a85d33f73823c7eb8af89a6e2f39925b/Tracking/src/Tracking/Reco/TruthSeedProcessor.cxx#L146](https://github.com/LDMX-Software/ldmx-sw/blob/25e45c03a85d33f73823c7eb8af89a6e2f39925b/Tracking/src/Tracking/Reco/TruthSeedProcessor.cxx#L146
)